nmakel/solaredge_meterproxy #3: Question: run the script on RPI in background

Maikel-K, 2020-11-28:
Hi Niels,

How do you run your script?
I tried running it via cron on my RPI but it fails for some reason.
I also tried to make a service of it but also this fails with a key error.

Any quick hints? Otherwise I will try and investigate some more.

Maikel

PS: thanks again for all your work! I love it thus far!

nmakel, 2020-11-28:
I use OpenWRT, so the most elegant way is a procd init script in /etc/init.d/.

It shouldn't be too difficult to make a systemd service. The most important thing is to make sure you use the `--config` parameter with the full path to the configuration file, and that the script is run as a user with sufficient privileges.

If you could paste the key error I might be able to figure out what is going on.

Maikel-K, 2020-11-28:
Im not at home but via SSH this is what I'm trying to achieve now true cron:

reboot python3 /home/pi/eastronsolaredge/semp-rtu.py

But this doesn't seem to work.

Service file content:

[Unit]                                                                           Description=Eastron-to-Solaredge                                                 After=multi-user.target                                                                                                                                           [Service]                                                                        ExecStart=/usr/bin/python3 /home/pi/eastronsolaredge/semp-rtu.py                 Restart=always                                                                                                                                                    [Install]                                                                        WantedBy=multi-user.target   

This is the output of the service:

 eastronsolaredge.service - Eastron-to-Solaredge                                   Loaded: loaded (/lib/systemd/system/eastronsolaredge.service; enabled; vendor preset: ena   Active: failed (Result: exit-code) since Sat 2020-11-28 18:25:53 CET; 4s ago                                                                        Process: 14319 ExecStart=/usr/bin/python3 /home/pi/eastronsolaredge/semp-rtu.py (code=exit                                                                      Main PID: 14319 (code=exited, status=1/FAILURE)                                                                                                                  Nov 28 18:25:53 raspberrypi4 systemd[1]: eastronsolaredge.service: Service RestartSec=100ms Nov 28 18:25:53 raspberrypi4 systemd[1]: eastronsolaredge.service: Scheduled restart job, reNov 28 18:25:53 raspberrypi4 systemd[1]: Stopped Eastron-to-Solaredge.                                                                   Nov 28 18:25:53 raspberrypi4 systemd[1]: eastronsolaredge.service: Start request repeated to                                                                     Nov 28 18:25:53 raspberrypi4 systemd[1]: eastronsolaredge.service: Failed with result 'exit-                                                                     Nov 28 18:25:53 raspberrypi4 systemd[1]: Failed to start Eastron-to-Solaredge.   ~                                                                                ~                                       

Maikel-K, 2020-11-28:
Sorry thats a bit messy I'm not at home and looked this up via iTerminal app....

nmakel, 2020-11-28:
You need to tell semp-rtu.py where your config is, either by setting a working directory in the service file, or by passing the '--config' parameter.

Maikel-K, 2020-11-28:
So you mean:

ExecStart=/usr/bin/python3 /home/pi/eastronsolaredge/semp-rtu.py --config semp.conf

Or 

ExecStart=/usr/bin/python3 /home/pi/eastronsolaredge/semp-rtu.py --config /Home/pi/eastronsolaredge/semp.conf

nmakel, 2020-11-28:
The last one.

Maikel-K, 2020-11-28:
Okay, will try that later on.
Cant we fix that in de main file that it will look for it? Just a thought?

Maikel-K, 2020-11-28:
So I fixed a few typos on my phone and the script seems to be called correctly with the config file loaded and now I get this,
It can't load sdm_modbus

[image]

nmakel, 2020-11-29:
The environment a script started by systemd runs in is different from that when run in a user session. You need to make sure all the dependencies are installed globally, or that the script runs as the correct user.

Maikel-K, 2020-11-29:
Easy as pi (pun intended), your input an the user addition runs the service:

Here is my service file content:
[Unit]
Description=Eastron-to-Solaredge
After=multi-user.target

[Service]
Type=idle
ExecStart=/usr/bin/python3 semp-rtu.py --config semp.conf
WorkingDirectory=/home/pi/eastronsolaredge/
User=pi

[Install]
WantedBy=multi-user.target

closed 2020-12-24 by nmakel