nmakel/solaredge_meterproxy #5: semp-tcp.py integer error for port number

Maikel-K, 2020-12-14:
Hi Niels,

I know you made this semp-tcp version for my request and couldnt test it yourself.
I recieved the gateways and am testing it now.

The Meter is working via ModbusTCP, i have run the script with semp.rtu.py fine for the last couple of days.

I get this error with the script:

pi@raspberrypi4:~/eastronsolaredgetcp $ python3 semp-tcp.py
2020-12-14 14:43:59 INFO: Created <Thread(t_update_2, initial)>: meter1 sdm630 SDM630(192.168.1.108:8899, connectionType.TCP: timeout=1, retries=3, unit=0x1)
2020-12-14 14:44:00 INFO: Starting <Thread(t_update_2, started 3053270112)>
Traceback (most recent call last):
  File "semp-tcp.py", line 251, in <module>
    confparser["server"].get("port", fallback=default_config["server"]["port"])
  File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/server/sync.py", line 616, in StartTcpServer
    server = ModbusTcpServer(context, framer, identity, address, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/server/sync.py", line 339, in __init__
    self.handler)
  File "/usr/lib/python3.7/socketserver.py", line 452, in __init__
    self.server_bind()
  File "/usr/lib/python3.7/socketserver.py", line 466, in server_bind
    self.socket.bind(self.server_address)
TypeError: an integer is required (got type str)

Maikel-K, 2020-12-14:
Oh, BTW this is my config

[server]
address = 192.168.1.115
port = 8898
meters = meter1
log_level = INFO

[meter1]
dst_address = 2
src_address = 1
type = sdm630
host = 192.168.1.108
port = 8899

nmakel, 2020-12-14:
Great to hear you're trying it out. The TCP port is now explicitly cast to integer. Give it another try.

Maikel-K, 2020-12-14:
python3 semp-tcp.py
2020-12-14 15:37:18 INFO: Created <Thread(t_update_2, initial)>: meter1 sdm630 SDM630(192.168.1.108:8899, connectionType.TCP: timeout=1, retries=3, unit=0x1)
2020-12-14 15:37:19 INFO: Starting <Thread(t_update_2, started 3053585504)>
Traceback (most recent call last):
  File "semp-tcp.py", line 251, in <module>
    confparser["server"].getint("port", fallback=default_config["server"]["port"])
  File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/server/sync.py", line 616, in StartTcpServer
    server = ModbusTcpServer(context, framer, identity, address, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/pymodbus/server/sync.py", line 339, in __init__
    self.handler)
  File "/usr/lib/python3.7/socketserver.py", line 452, in __init__
    self.server_bind()
  File "/usr/lib/python3.7/socketserver.py", line 466, in server_bind
    self.socket.bind(self.server_address)
OSError: [Errno 99] Cannot assign requested address

nmakel, 2020-12-14:
Is 192.168.1.115 the IP of your Pi, and is it assigned to an interface? Can you ping the address?

Maikel-K, 2020-12-14:
192.168.1.115 is ip of the gateway connected to the solaredge.

nmakel, 2020-12-14:
In that case you need to use the IP adress of the Pi, not the gateway. In turn you will need to tell the gateway to be a TCP client and provide it with the IP of the Pi.

Maikel-K, 2020-12-14:
So my config is now:
address = 192.168.1.104
port = 5502

 i should tell my gateway to:
server: 192.168.1.104
port: 5502

nmakel, 2020-12-14:
Correct. If you then also set log_level to debug, hopefully you will see the SolarEdge talking to your Pi.

Maikel-K, 2020-12-14:
python3 semp-tcp.py
2020-12-14 16:35:53 DEBUG: [1234, 0, 5, 5, 5, 5, 0, 0, 0, 15, 1, 10000, 10000, 10000, 64536, 64536, 64536, 1500, 120, 0, 0, 20000, 0]
2020-12-14 16:35:53 DEBUG: setValues[3] 1601:23
2020-12-14 16:35:53 DEBUG: [0, 2, 4, 0, 0, 5]
2020-12-14 16:35:53 DEBUG: setValues[3] 1651:6
2020-12-14 16:35:53 DEBUG: [4614, 15, 0, 0, 0, 0, 0, 25, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2020-12-14 16:35:53 DEBUG: setValues[3] 1701:23
2020-12-14 16:35:53 INFO: Created <Thread(t_update_2, initial)>: meter1 sdm630 SDM630(192.168.1.108:8899, connectionType.TCP: timeout=1, retries=3, unit=0x1)
2020-12-14 16:35:54 INFO: Starting <Thread(t_update_2, started 3052926048)>
2020-12-14 16:35:54 DEBUG: Current transaction state - IDLE
2020-12-14 16:35:54 DEBUG: Running transaction 1
2020-12-14 16:35:54 DEBUG: SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x50
2020-12-14 16:35:54 DEBUG: New Transaction state 'SENDING'
2020-12-14 16:35:54 DEBUG: Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2020-12-14 16:35:55 DEBUG: Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2020-12-14 16:35:55 DEBUG: RECV: 0x0 0x1 0x0 0x0 0x0 0xa3 0x1 0x4 0xa0 0x43 0x63 0xf8 0x39 0x43 0x66 0x23 0x5 0x43 0x67 0xd1 0x39 0x3f 0x78 0xe1 0xf0 0x3f 0x88 0x82 0xdb 0x3f 0x3e 0x47 0x97 0x43 0x57 0x24 0xee 0x43 0x5c 0x92 0x76 0x43 0x5 0x3b 0x6e 0x43 0x5d 0xc4 0xa0 0x43 0x75 0x6d 0x23 0x43 0x2c 0x4a 0x28 0x42 0x57 0x2f 0x4f 0xc2 0xd7 0x3d 0x5a 0x42 0xda 0x79 0xd6 0x3f 0x78 0x5a 0x8d 0x3f 0x66 0x13 0x32 0x3f 0x45 0xf7 0x30 0x41 0x60 0xa3 0x24 0xc1 0xd0 0x14 0xf5 0x42 0x1d 0x67 0xe6 0x43 0x65 0xf8 0x63 0x0 0x0 0x0 0x0 0x3f 0x6d 0x70 0x4c 0x40 0x32 0x14 0x39 0x0 0x0 0x0 0x0 0x44 0xe 0x3c 0xb4 0x0 0x0 0x0 0x0 0x44 0xe 0xe9 0x1 0x0 0x0 0x0 0x0 0x42 0x5d 0xa8 0x47 0x3f 0x7e 0xcb 0x58 0x0 0x0 0x0 0x0 0x40 0xb2 0x7 0x1c 0x0 0x0 0x0 0x0 0x42 0x47 0x9e 0xd8 0x43 0xfe 0x66 0xc8 0x42 0x7b 0xb8 0x51 0x42 0x90 0x11 0x68 0x42 0xfa 0xd7 0x8d
2020-12-14 16:35:55 DEBUG: Processing: 0x0 0x1 0x0 0x0 0x0 0xa3 0x1 0x4 0xa0 0x43 0x63 0xf8 0x39 0x43 0x66 0x23 0x5 0x43 0x67 0xd1 0x39 0x3f 0x78 0xe1 0xf0 0x3f 0x88 0x82 0xdb 0x3f 0x3e 0x47 0x97 0x43 0x57 0x24 0xee 0x43 0x5c 0x92 0x76 0x43 0x5 0x3b 0x6e 0x43 0x5d 0xc4 0xa0 0x43 0x75 0x6d 0x23 0x43 0x2c 0x4a 0x28 0x42 0x57 0x2f 0x4f 0xc2 0xd7 0x3d 0x5a 0x42 0xda 0x79 0xd6 0x3f 0x78 0x5a 0x8d 0x3f 0x66 0x13 0x32 0x3f 0x45 0xf7 0x30 0x41 0x60 0xa3 0x24 0xc1 0xd0 0x14 0xf5 0x42 0x1d 0x67 0xe6 0x43 0x65 0xf8 0x63 0x0 0x0 0x0 0x0 0x3f 0x6d 0x70 0x4c 0x40 0x32 0x14 0x39 0x0 0x0 0x0 0x0 0x44 0xe 0x3c 0xb4 0x0 0x0 0x0 0x0 0x44 0xe 0xe9 0x1 0x0 0x0 0x0 0x0 0x42 0x5d 0xa8 0x47 0x3f 0x7e 0xcb 0x58 0x0 0x0 0x0 0x0 0x40 0xb2 0x7 0x1c 0x0 0x0 0x0 0x0 0x42 0x47 0x9e 0xd8 0x43 0xfe 0x66 0xc8 0x42 0x7b 0xb8 0x51 0x42 0x90 0x11 0x68 0x42 0xfa 0xd7 0x8d
2020-12-14 16:35:55 DEBUG: Factory Response[ReadInputRegistersResponse: 4]
2020-12-14 16:35:55 DEBUG: Adding transaction 1
2020-12-14 16:35:55 DEBUG: Getting transaction 1
2020-12-14 16:35:55 DEBUG: Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2020-12-14 16:35:55 DEBUG: [17251, 63545, 17254, 8965, 17255, 53561, 16248, 57840, 16264, 33499, 16190, 18327, 17239, 9454, 17244, 37494, 17157, 15214, 17245, 50336, 17269, 27939, 17196, 18984, 16983, 12111, 49879, 15706, 17114, 31190, 16248, 23181, 16230, 4914, 16197, 63280, 16736, 41764, 49616, 5365, 16925, 26598, 17253, 63587, 0, 0, 16237, 28748, 16434, 5177, 0, 0, 17422, 15540, 0, 0, 17422, 59649, 0, 0, 16989, 43079, 16254, 52056, 0, 0, 16562, 1820, 0, 0, 16967, 40664, 17406, 26312, 17019, 47185, 17040, 4456, 17146, 55181]
2020-12-14 16:35:55 DEBUG: [b'Cc', b'\xf89']
2020-12-14 16:35:55 DEBUG: [b'Cf', b'#\x05']
2020-12-14 16:35:55 DEBUG: [b'Cg', b'\xd19']
2020-12-14 16:35:55 DEBUG: [b'?x', b'\xe1\xf0']
2020-12-14 16:35:55 DEBUG: [b'?\x88', b'\x82\xdb']
2020-12-14 16:35:55 DEBUG: [b'?>', b'G\x97']
2020-12-14 16:35:55 DEBUG: [b'CW', b'$\xee']
2020-12-14 16:35:55 DEBUG: [b'C\\', b'\x92v']
2020-12-14 16:35:55 DEBUG: [b'C\x05', b';n']
2020-12-14 16:35:55 DEBUG: [b'C]', b'\xc4\xa0']
2020-12-14 16:35:55 DEBUG: [b'Cu', b'm#']
2020-12-14 16:35:55 DEBUG: [b'C,', b'J(']
2020-12-14 16:35:55 DEBUG: [b'BW', b'/O']
2020-12-14 16:35:55 DEBUG: [b'\xc2\xd7', b'=Z']
2020-12-14 16:35:55 DEBUG: [b'B\xda', b'y\xd6']
2020-12-14 16:35:55 DEBUG: [b'?x', b'Z\x8d']
2020-12-14 16:35:55 DEBUG: [b'?f', b'\x132']
2020-12-14 16:35:55 DEBUG: [b'?E', b'\xf70']
2020-12-14 16:35:55 DEBUG: [b'A`', b'\xa3$']
2020-12-14 16:35:55 DEBUG: [b'\xc1\xd0', b'\x14\xf5']
2020-12-14 16:35:55 DEBUG: [b'B\x1d', b'g\xe6']
2020-12-14 16:35:55 DEBUG: [b'Ce', b'\xf8c']
2020-12-14 16:35:55 DEBUG: [b'?m', b'pL']
2020-12-14 16:35:55 DEBUG: [b'@2', b'\x149']
2020-12-14 16:35:55 DEBUG: [b'D\x0e', b'<\xb4']
2020-12-14 16:35:55 DEBUG: [b'D\x0e', b'\xe9\x01']
2020-12-14 16:35:55 DEBUG: [b'B]', b'\xa8G']
2020-12-14 16:35:55 DEBUG: [b'?~', b'\xcbX']
2020-12-14 16:35:55 DEBUG: [b'@\xb2', b'\x07\x1c']
2020-12-14 16:35:55 DEBUG: [b'BG', b'\x9e\xd8']
2020-12-14 16:35:55 DEBUG: [b'C\xfe', b'f\xc8']
2020-12-14 16:35:55 DEBUG: [b'B{', b'\xb8Q']
2020-12-14 16:35:55 DEBUG: [b'B\x90', b'\x11h']
2020-12-14 16:35:55 DEBUG: [b'B\xfa', b'\xd7\x8d']
2020-12-14 16:35:55 DEBUG: Current transaction state - TRANSACTION_COMPLETE
2020-12-14 16:35:55 DEBUG: Running transaction 2
2020-12-14 16:35:55 DEBUG: SEND: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x50 0x0 0x1c
2020-12-14 16:35:55 DEBUG: New Transaction state 'SENDING'
[... cut ...]
2020-12-14 16:36:01 DEBUG: [b'D\x0f', b'\x11Y']
2020-12-14 16:36:01 DEBUG: [b'B^', b'z\xe9']
2020-12-14 16:36:01 DEBUG: [b'?~', b'\xc9\xbb']
2020-12-14 16:36:01 DEBUG: [b'@\xb2', b"~'"]
2020-12-14 16:36:01 DEBUG: [b'BG', b'\xc9\xbc']
2020-12-14 16:36:01 DEBUG: [b'C\xfe', b'f\xc8']
2020-12-14 16:36:01 DEBUG: [b'B{', b'\xb8Q']
2020-12-14 16:36:01 DEBUG: [b'B\x90', b'\x11h']
2020-12-14 16:36:01 DEBUG: [b'B\xfa', b'\xd7\x8d']
2020-12-14 16:36:01 DEBUG: Current transaction state - TRANSACTION_COMPLETE

Maikel-K, 2020-12-14:
So it seems to run, but no data in the solaredge app?
Any clues?

Maikel-K, 2020-12-14:
I see this coming buy:
2020-12-14 16:35:55 DEBUG: Started thread to serve client at ('192.168.1.115', 7506)
2020-12-14 16:35:55 DEBUG: Client Connected [192.168.1.115:7506]

With, what it looks like, a changing port all the time?

Maikel-K, 2020-12-14:
I checked on the inverter itself. It says com error, missing meter.

So it doesn't seem to work? What could be wrong?

nmakel, 2020-12-14:
Could you provide a screenshot of the gateway setup screen, specifically where you set the server IP, port, and maybe some other options? As far as I can see you are receiving requests from the inverter, `Factory Response[ReadInputRegistersResponse: 4]` for example.

Maikel-K, 2020-12-14:
[image]
[image]

Maikel-K, 2020-12-14:
i did once again checked if the connection between PI-->Gateway-->inverter is working with switchting back to non-solaredge-logger and using the solaredge-modbus script.
The connection is working, so the gateway working as SERVER works okay!

nmakel, 2020-12-14:
Turns out the modbus messages you're seeing are caused by the meter updating values in semp-tcp.py, not the inverter sending requests. I will need to look into the documentation of your gateway a bit more to see what settings are required.

Maikel-K, 2020-12-14:
Hi Niels,

Im using this unit: EE11

http://www.hi-flying.com/elfin-ee10-elfin-ee11

nmakel, 2020-12-14:
The documentation is frustratingly incoherent. Would you mind setting the gateway back to tcp client, with all the settings back to how you had them, and then put as many screenshots of the system, serial port, and socket settings pages. Perhaps it's as simple as a single option set incorrectly.

Maikel-K, 2020-12-14:
Hi Niels, 

These are the settings for the serial port. Not much more to be set actually...
[image]
[image]

nmakel, 2020-12-14:
Could you set the baud rate to 9600?

Also, what options are available in the Protocol dropdown?

Maikel-K, 2020-12-14:
Like i write before, the serial to tcp server part works as I can retrieve inverter data with your Solaredge-modbus scripts.

Maikel-K, 2020-12-14:
Our messages crossed each other.
These are the protocol options.

Maybe it should be None

[image]

nmakel, 2020-12-14:
No, if I understand the documentation correctly the Modbus protocol setting enables Modbus RTU to TCP translation.

I understand the Inverter acting as server works (gateway is client to the inverter modbus rtu server). But in this case the inverter must act as client, and it may be more particular regarding serial settings.

Maikel-K, 2020-12-14:
Would it be possible to write a simple script to see if the inverter is sending/requesting data only and no info of the meter?

This way we can debug the connection to the meter?

nmakel, 2020-12-14:
> Would it be possible to write a simple script to see if the inverter is sending/requesting data only and no info of the meter?
> 
> This way we can debug the connection to the meter?

Sure. Have you tried a baud rate of 9600 already?

Maikel-K, 2020-12-14:
Yes i tried. No difference as in I don't see any data appearing in SolarEdge.

I put my laptop away and copying data from the tool on my iPhone is very messy....

nmakel, 2020-12-15:
Try the following script. It creates a simple Modbus TCP server, and has logging set to debug.

```
#!/usr/bin/env python3

import logging
import sys

from pymodbus.server.sync import StartTcpServer
from pymodbus.constants import Endian
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.transaction import ModbusRtuFramer
from pymodbus.datastore import ModbusSlaveContext
from pymodbus.datastore import ModbusServerContext
from pymodbus.payload import BinaryPayloadBuilder

log_handler = logging.StreamHandler(sys.stdout)
log_handler.setFormatter(logging.Formatter("%(asctime)s %(levelname)s: %(message)s", datefmt="%Y-%m-%d %H:%M:%S"))

logger = logging.getLogger()
logger.addHandler(log_handler)
logger.setLevel(logging.DEBUG)

slave = ModbusSlaveContext()
block_1601 = BinaryPayloadBuilder(byteorder=Endian.Big, wordorder=Endian.Little)
block_1601.add_32bit_int(1234) # config passcode
slave.setValues(3, 1600, block_1601.to_registers())

identity = ModbusDeviceIdentification()
server_ctx = ModbusServerContext(slaves={"2": slave}, single=False)

StartTcpServer(
    server_ctx,
    identity=identity,
    address=("192.168.1.104", 5502)
)
```

Maikel-K, 2020-12-15:
How fast should i start to see data coming in Putty?

pi@raspberrypi4:~/eastronsolaredgetcp $ python3 test.py
2020-12-15 13:47:32 DEBUG: [1234, 0]
2020-12-15 13:47:32 DEBUG: setValues[3] 1601:2
2020-12-15 13:47:34 DEBUG: Started thread to serve client at ('192.168.1.115', 3561)
2020-12-15 13:47:34 DEBUG: Client Connected [192.168.1.115:3561]

In the gateway at communication settings i can also set the local port. If i set it to 5502 this port is also shown in putty.

Maikel-K, 2020-12-15:
Hi, so the output now is:

pi@raspberrypi4:~/eastronsolaredgetcp $ python3 test.py
2020-12-15 13:47:32 DEBUG: [1234, 0]
2020-12-15 13:47:32 DEBUG: setValues[3] 1601:2
2020-12-15 13:47:34 DEBUG: Started thread to serve client at ('192.168.1.115', 3                                                                                              561)
2020-12-15 13:47:34 DEBUG: Client Connected [192.168.1.115:3561]

Maikel-K, 2020-12-15:
i changed at Serial port settings Protocol to NONE and this immediatly happened.
I also saw data the data counter started to increase on the TCP-client part of the status page

2020-12-15 14:26:31 DEBUG: Frame check failed, ignoring!!
2020-12-15 14:26:33 DEBUG: Handling data: 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2020-12-15 14:26:33 DEBUG: Processing: 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2020-12-15 14:26:33 DEBUG: Frame check failed, ignoring!!
2020-12-15 14:26:35 DEBUG: Handling data: 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2020-12-15 14:26:35 DEBUG: Processing: 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2020-12-15 14:26:35 DEBUG: Frame check failed, ignoring!!
2020-12-15 14:26:37 DEBUG: Handling data: 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2020-12-15 14:26:37 DEBUG: Processing: 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2020-12-15 14:26:37 DEBUG: Frame check failed, ignoring!!

nmakel, 2020-12-15:
> 2020-12-15 13:47:32 DEBUG: [1234, 0]
> 2020-12-15 13:47:32 DEBUG: setValues[3] 1601:2

This is a part of the script, one of the registers is being filled with dummy data.

> 2020-12-15 13:47:34 DEBUG: Started thread to serve client at ('192.168.1.115', 3 561)
> 2020-12-15 13:47:34 DEBUG: Client Connected [192.168.1.115:3561]

This is your gateway connecting with the Pi. Immediately after this message you should be seeing a large number of requests every second. They look like this:

> 2020-12-14 17:00:28 DEBUG: Getting Frame - 0x3 0x3 0xf2 0x0 0x6
> 2020-12-14 17:00:28 DEBUG: Factory Request[ReadHoldingRegistersRequest: 3]
> 2020-12-14 17:00:28 DEBUG: Frame advanced, resetting header!!
> 2020-12-14 17:00:28 DEBUG: validate: fc-[3] address-1011: count-6
> 2020-12-14 17:00:28 DEBUG: getValues fc-[3] address-1011: count-6
> 2020-12-14 17:00:28 DEBUG: send: [ReadRegisterResponse (6)]- b'02030c999a439000000000000000008deb'

You're not, so either **the gateway is not forwarding messages** from your inverter, or **the inverter is not sending the messages**. Both could be a configuration issue in the gateway or inverter.

Remove the gateway and connect the Pi your inverter directly using the RS485 to USB adapter and run semp-rtu.py. This way you can rule out the inverter being the problem.

Maikel-K, 2020-12-15:
Immediately the proxy is running again.

nmakel, 2020-12-15:
> Immediately the proxy is running again.

Ok, great. Then we can ignore the inverter.

To rule out a networking issue, are you blocking traffic with a firewall on your Pi? If so, be sure to turn it off.

I am also curious what the test script shows when you choose frame as protocol.

Maikel-K, 2020-12-15:
Do you have a clue what we need to set here then:
[image]

nmakel, 2020-12-15:
> I didn't install a firewall myself on the pin so I don't think so

Run `sudo iptables -L` to check.

> Do you have a clue what we need to set here then, I can switch tag to off too.

No clue, might only be useful if the data is known beforehand. Try turning it off?

Maikel-K, 2020-12-15:
pi@raspberrypi4:~$ sudo iptables -L                                                          Chain INPUT (policy ACCEPT)                                                                  target     prot opt source               destination                                                                                                                                      Chain FORWARD (policy ACCEPT)                                                                target     prot opt source               destination                                                                                                                                      Chain OUTPUT (policy ACCEPT)                                                                 target     prot opt source               destination

Maikel-K, 2020-12-15:
Dont pay attention to the black question marks...... those are spaces

nmakel, 2020-12-16:
Firewall is fine, not blocking anything. I guess it's just a matter of finding the correct configuration on your gateway.

You should be able to see things are working right away using the test script. It may also be useful to use the acompanying desktop utility to check whether the gateway is actually receiving messages. Remember that the inverter might need different serial settings depending on whether it is set to non-solaredge logger (inverter as server) or modbus multi-devices (inverter as client). In my experience it will use baud 9600 when it is a modbus client.

Maikel-K, 2020-12-16:
its now showing this: 

2020-12-16 08:50:21 DEBUG: Handling data: 0x2 0x3 0x6 0x40 0x0 0x17 0x4 0xab
2020-12-16 08:50:21 DEBUG: Processing: 0x2 0x3 0x6 0x40 0x0 0x17 0x4 0xab
2020-12-16 08:50:21 DEBUG: Frame check failed, ignoring!!
2020-12-16 08:50:23 DEBUG: Handling data: 0x2 0x3 0x6 0x40 0x0 0x17 0x4 0xab
2020-12-16 08:50:23 DEBUG: Processing: 0x2 0x3 0x6 0x40 0x0 0x17 0x4 0xab
2020-12-16 08:50:23 DEBUG: Frame check failed, ignoring!!
2020-12-16 08:50:25 DEBUG: Handling data: 0x2 0x3 0x6 0x40 0x0 0x17 0x4 0xab
2020-12-16 08:50:25 DEBUG: Processing: 0x2 0x3 0x6 0x40 0x0 0x17 0x4 0xab
2020-12-16 08:50:25 DEBUG: Frame check failed, ignoring!!

nmakel, 2020-12-16:
> its now showing this:
>
> 2020-12-16 08:50:21 DEBUG: Handling data: 0x2 0x3 0x6 0x40 0x0 0x17 0x4 0xab
> 2020-12-16 08:50:21 DEBUG: Processing: 0x2 0x3 0x6 0x40 0x0 0x17 0x4 0xab
> 2020-12-16 08:50:21 [...]

Excellent, this is progress. I see similar messages. Have you tried the original semp-tcp.py?

Maikel-K, 2020-12-16:
Hi Niels,

i checked it, following is coming by.
I checked the solaredge app, and i have seen that there where values above the house icon and the grid was showing 0
0,39 kw production, 0.39 kw load(house) 0.0kw export.

So i think the script is running, but now the app isnt updating anymore but the website isnt updating too so maybe something is going on at the SolarEdge server at the moment.

pi@raspberrypi4:~/eastronsolaredgetcp $ python3 semp-tcp.py
2020-12-16 13:47:30 DEBUG: [1234, 0, 5, 5, 5, 5, 0, 0, 0, 15, 1, 10000, 10000, 10000, 64536, 64536, 64536, 1500, 120, 0, 0, 20000, 0]
2020-12-16 13:47:30 DEBUG: setValues[3] 1601:23
2020-12-16 13:47:30 DEBUG: [0, 2, 4, 0, 0, 5]
2020-12-16 13:47:30 DEBUG: setValues[3] 1651:6
2020-12-16 13:47:30 DEBUG: [4614, 15, 0, 0, 0, 0, 0, 25, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2020-12-16 13:47:30 DEBUG: setValues[3] 1701:23
2020-12-16 13:47:30 INFO: Created <Thread(t_update_2, initial)>: meter1 sdm630 SDM630(192.168.1.108:8899, connectionType.TCP: timeout=1, retries=3, unit=0x1)
2020-12-16 13:47:31 INFO: Starting <Thread(t_update_2, started 3053786208)>
2020-12-16 13:47:31 DEBUG: Current transaction state - IDLE
2020-12-16 13:47:31 DEBUG: Running transaction 1
2020-12-16 13:47:31 DEBUG: SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x50
2020-12-16 13:47:31 DEBUG: New Transaction state 'SENDING'
2020-12-16 13:47:31 DEBUG: Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2020-12-16 13:47:31 DEBUG: Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2020-12-16 13:47:31 DEBUG: RECV: 0x0 0x1 0x0 0x0 0x0 0xa3 0x1 0x4 0xa0 0x43 0x6a 0x81 0x69 0x43 0x6a 0x7c 0x13 0x43 0x68 0x9a 0x3f 0x3f 0x11 0x67 0x50 0x3f 0x87 0x5e 0x4c 0x3e 0x7c 0xc 0x2d 0x43 0x2 0x73 0x89 0xc3 0x6e 0xa2 0xff 0x42 0x5e 0x53 0x6b 0x43 0x5 0xa1 0x6c 0x43 0x77 0x59 0x41 0x42 0x63 0xcc 0xe7 0x41 0xe7 0xce 0x14 0xc2 0x82 0x22 0xf9 0xc1 0x46 0x90 0x8a 0x3f 0x79 0xe8 0xce 0xbf 0x76 0xfb 0xbc 0x3f 0x79 0xd9 0x6 0x41 0x48 0x62 0x80 0x43 0x43 0x40 0xc5 0xc1 0x49 0x66 0x7f 0x43 0x69 0xf0 0x50 0x0 0x0 0x0 0x0 0x3f 0x1f 0x7c 0x1f 0x3f 0xef 0x3a 0x2f 0x0 0x0 0x0 0x0 0xc2 0x52 0x6a 0x6d 0x0 0x0 0x0 0x0 0x42 0x8f 0x1a 0xb4 0x0 0x0 0x0 0x0 0xc2 0x42 0x3 0xa 0xbf 0x3c 0x34 0xf4 0x0 0x0 0x0 0x0 0x43 0x5e 0xae 0x33 0x0 0x0 0x0 0x0 0x42 0x47 0xc8 0x86 0x44 0x5 0xf7 0x5c 0x42 0x7f 0xe3 0x53 0x42 0x96 0xdd 0x2f 0x43 0x3 0x96 0x45
2020-12-16 13:47:31 DEBUG: Processing: 0x0 0x1 0x0 0x0 0x0 0xa3 0x1 0x4 0xa0 0x43 0x6a 0x81 0x69 0x43 0x6a 0x7c 0x13 0x43 0x68 0x9a 0x3f 0x3f 0x11 0x67 0x50 0x3f 0x87 0x5e 0x4c 0x3e 0x7c 0xc 0x2d 0x43 0x2 0x73 0x89 0xc3 0x6e 0xa2 0xff 0x42 0x5e 0x53 0x6b 0x43 0x5 0xa1 0x6c 0x43 0x77 0x59 0x41 0x42 0x63 0xcc 0xe7 0x41 0xe7 0xce 0x14 0xc2 0x82 0x22 0xf9 0xc1 0x46 0x90 0x8a 0x3f 0x79 0xe8 0xce 0xbf 0x76 0xfb 0xbc 0x3f 0x79 0xd9 0x6 0x41 0x48 0x62 0x80 0x43 0x43 0x40 0xc5 0xc1 0x49 0x66 0x7f 0x43 0x69 0xf0 0x50 0x0 0x0 0x0 0x0 0x3f 0x1f 0x7c 0x1f 0x3f 0xef 0x3a 0x2f 0x0 0x0 0x0 0x0 0xc2 0x52 0x6a 0x6d 0x0 0x0 0x0 0x0 0x42 0x8f 0x1a 0xb4 0x0 0x0 0x0 0x0 0xc2 0x42 0x3 0xa 0xbf 0x3c 0x34 0xf4 0x0 0x0 0x0 0x0 0x43 0x5e 0xae 0x33 0x0 0x0 0x0 0x0 0x42 0x47 0xc8 0x86 0x44 0x5 0xf7 0x5c 0x42 0x7f 0xe3 0x53 0x42 0x96 0xdd 0x2f 0x43 0x3 0x96 0x45
2020-12-16 13:47:31 DEBUG: Factory Response[ReadInputRegistersResponse: 4]
2020-12-16 13:47:31 DEBUG: Adding transaction 1
2020-12-16 13:47:31 DEBUG: Getting transaction 1
2020-12-16 13:47:31 DEBUG: Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2020-12-16 13:47:31 DEBUG: [17258, 33129, 17258, 31763, 17256, 39487, 16145, 26448, 16263, 24140, 15996, 3117, 17154, 29577, 50030, 41727, 16990, 21355, 17157, 41324, 17271, 22849, 16995, 52455, 16871, 52756, 49794, 8953, 49478, 37002, 16249, 59598, 49014, 64444, 16249, 55558, 16712, 25216, 17219, 16581, 49481, 26239, 17257, 61520, 0, 0, 16159, 31775, 16367, 14895, 0, 0, 49746, 27245, 0, 0, 17039, 6836, 0, 0, 49730, 778, 48956, 13556, 0, 0, 17246, 44595, 0, 0, 16967, 51334, 17413, 63324, 17023, 58195, 17046, 56623, 17155, 38469]
2020-12-16 13:47:31 DEBUG: [b'Cj', b'\x81i']
2020-12-16 13:47:31 DEBUG: [b'Cj', b'|\x13']
2020-12-16 13:47:31 DEBUG: [b'Ch', b'\x9a?']
2020-12-16 13:47:31 DEBUG: [b'?\x11', b'gP']
2020-12-16 13:47:31 DEBUG: [b'?\x87', b'^L']
2020-12-16 13:47:31 DEBUG: [b'>|', b'\x0c-']
2020-12-16 13:47:31 DEBUG: [b'C\x02', b's\x89']
2020-12-16 13:47:31 DEBUG: [b'\xc3n', b'\xa2\xff']
2020-12-16 13:47:31 DEBUG: [b'B^', b'Sk']
2020-12-16 13:47:31 DEBUG: [b'C\x05', b'\xa1l']
2020-12-16 13:47:31 DEBUG: [b'Cw', b'YA']
2020-12-16 13:47:31 DEBUG: [b'Bc', b'\xcc\xe7']
2020-12-16 13:47:31 DEBUG: [b'A\xe7', b'\xce\x14']
2020-12-16 13:47:31 DEBUG: [b'\xc2\x82', b'"\xf9']
2020-12-16 13:47:31 DEBUG: [b'\xc1F', b'\x90\x8a']
2020-12-16 13:47:31 DEBUG: [b'?y', b'\xe8\xce']
2020-12-16 13:47:31 DEBUG: [b'\xbfv', b'\xfb\xbc']
2020-12-16 13:47:31 DEBUG: [b'?y', b'\xd9\x06']
2020-12-16 13:47:31 DEBUG: [b'AH', b'b\x80']
2020-12-16 13:47:31 DEBUG: [b'CC', b'@\xc5']
2020-12-16 13:47:31 DEBUG: [b'\xc1I', b'f\x7f']
2020-12-16 13:47:31 DEBUG: [b'Ci', b'\xf0P']
2020-12-16 13:47:31 DEBUG: [b'?\x1f', b'|\x1f']
2020-12-16 13:47:31 DEBUG: [b'?\xef', b':/']
2020-12-16 13:47:31 DEBUG: [b'\xc2R', b'jm']
2020-12-16 13:47:31 DEBUG: [b'B\x8f', b'\x1a\xb4']
2020-12-16 13:47:31 DEBUG: [b'\xc2B', b'\x03\n']
2020-12-16 13:47:31 DEBUG: [b'\xbf<', b'4\xf4']
2020-12-16 13:47:31 DEBUG: [b'C^', b'\xae3']
2020-12-16 13:47:31 DEBUG: [b'BG', b'\xc8\x86']
[... cut ...]
2020-12-16 13:47:38 DEBUG: [b'\xc2%', b'\xa7B']
2020-12-16 13:47:38 DEBUG: [b'\xbfI', b'\x98I']
2020-12-16 13:47:38 DEBUG: [b'C^', b'\xc0\xfc']
2020-12-16 13:47:38 DEBUG: [b'BG', b'\x9c\xbc']
2020-12-16 13:47:38 DEBUG: [b'D\x05', b'\xf7|']
2020-12-16 13:47:38 DEBUG: [b'B\x7f', b'\xe4Z']
2020-12-16 13:47:38 DEBUG: [b'B\x96', b'\xdd/']
2020-12-16 13:47:38 DEBUG: [b'C\x03', b'\x96E']
2020-12-16 13:47:38 DEBUG: Current transaction state - TRANSACTION_COMPLETE
2020-12-16 13:47:38 DEBUG: Running transaction 6

nmakel, 2020-12-16:
You're still getting `Frame check failed, ignoring!!` errors. This means the inverter messages are corrupt or invalid and not being processed. The gateway is forwarding *something*, but not quite correctly. Are there any options set that *add* tags or something similar to the messages?

You're seeing `0x2 0x3 0x6 0x40 0x0 0x17 0x4 0xab`, where `0x2` is the address, `0x3` is read holding registers, and so on. So it looks like a valid message, but there is stuff added to the end which doesn't make sense. 

You will need to switch back to the test script and fiddle with the gateway options until you stop getting the error and requests more like this:

 ```
2020-12-14 17:00:28 DEBUG: Getting Frame - 0x3 0x3 0xf2 0x0 0x6
2020-12-14 17:00:28 DEBUG: Factory Request[ReadHoldingRegistersRequest: 3]
2020-12-14 17:00:28 DEBUG: Frame advanced, resetting header!!
```

Maikel-K, 2020-12-16:
So seeing data with 0x2 0x3 is at least good, now we need to fine tune like you said.....

Mm wonder what to check. I hope I have some time tonight!

Maikel-K, 2020-12-16:
I tried many different settings in the serial port part of the gateway but the only thing putting the data stream back to 0x0 data is the baud rate. changing data bits and stopbits are still giving 0x02 data etc.

nmakel, 2020-12-16:
The serial settings of the gateway should be baud 9600, 8 data bits, 1 stop bit, no parity. These are the settings of the device we want to emulate, the wattnode meter, which is what the solaredge inverter expects and will use to connect.

The documentation of this rs485-wifi gateway is rubbish -- there literally isn't a complete or correct english sentence in any of the documents. Not much I can recommend other than that you experiment with the various protocol and buffer settings.

The test script will show you when you've found the right combination.

Maikel-K, 2020-12-16:
would this make any sense? found it somewhere in a github post

from pymodbus.transaction import ModbusRtuFramer
client = ModbusClient('192.168.0.7', port=8899, framer=ModbusRtuFramer)

nmakel, 2020-12-16:
Give it a try, but as far as I know the framer property only works on serial clients and servers. You can try adding it to the `StartTcpServer` function in the test script.

Maikel-K, 2020-12-18:
Hi Niels,

So i didnt had a lot of time the last days but i want to inform you about the following:
If the RS485-Ethernet dongle is setup as server i can run the script to retrieve data from the inverter which in my opinion means the inverter is working and connected correctly. The issue is, as i can figure it out, the part when the inverter needs to be a client.

Is it possible to run the script below as "server" and then request the information again with the dongle as client?

 python3 example-tcp-inverteronly.py 192.168.1.115 8899
Inverter(192.168.1.115:8899, connectionType.TCP: timeout=1, retries=3, unit=0x1):

Registers:
        Manufacturer: SolarEdge
        Model: SE3500
        Type: Single Phase Inverter
        Version: 0003.2221
        Serial: 7316DD53
        Status: Sleeping
        Temperature: 20.77°C
        Current: 0.00A
        Voltage: 230.20V
        Frequency: 50.00Hz
        Power: 0.00W
        Power (Apparent): 0.00VA
        Power (Reactive): 0.00VAr
        Power Factor: 100.00%
        Total Energy: 7815914Wh
        DC Current: 0.00A
        DC Voltage: 0.00V
        DC Power: 0.00W

With script:

#!/usr/bin/env python3

import argparse
import json
import time

import solaredge_modbus

if __name__ == "__main__":
    argparser = argparse.ArgumentParser()
    argparser.add_argument("host", type=str, help="ModbusTCP address")
    argparser.add_argument("port", type=int, help="ModbusTCP port")
    argparser.add_argument("--timeout", type=int, default=1, help="Connection timeout")
    argparser.add_argument("--unit", type=int, default=1, help="Modbus unit")
    argparser.add_argument("--json", action="store_true", default=False, help="Output as JSON")
    args = argparser.parse_args()

    inverter = solaredge_modbus.Inverter(
        host=args.host,
        port=args.port,
        timeout=args.timeout,
        unit=args.unit
    )

    values = {}
    values = inverter.read_all()
    meters = inverter.meters()
    batteries = inverter.batteries()
    values["meters"] = {}
    values["batteries"] = {}

    for meter, params in meters.items():
        meter_values = params.read_all()
        values["meters"][meter] = meter_values

    if args.json:
        print(json.dumps(values, indent=4))
    else:
        print(f"{inverter}:")
        print("\nRegisters:")

        print(f"\tManufacturer: {values['c_manufacturer']}")
        print(f"\tModel: {values['c_model']}")
        print(f"\tType: {solaredge_modbus.C_SUNSPEC_DID_MAP[str(values['c_sunspec_did'])]}")
        print(f"\tVersion: {values['c_version']}")
        print(f"\tSerial: {values['c_serialnumber']}")
        print(f"\tStatus: {solaredge_modbus.INVERTER_STATUS_MAP[values['status']]}")
        print(f"\tTemperature: {(values['temperature'] * (10 ** values['temperature_scale'])):.2f}{inverter.registers['temperature'][6]}")

        print(f"\tCurrent: {(values['current'] * (10 ** values['temperature_scale'])):.2f}{inverter.registers['current'][6]}")

        if values['c_sunspec_did'] is solaredge_modbus.sunspecDID.THREE_PHASE_INVERTER:
            print(f"\tPhase 1 Current: {(values['p1_current'] * (10 ** values['current_scale'])):.2f}{inverter.registers['p1_current'][6]}")
            print(f"\tPhase 2 Current: {(values['p2_current'] * (10 ** values['current_scale'])):.2f}{inverter.registers['p2_current'][6]}")
            print(f"\tPhase 3 Current: {(values['p3_current'] * (10 ** values['current_scale'])):.2f}{inverter.registers['p3_current'][6]}")
            print(f"\tPhase 1 voltage: {(values['p1_voltage'] * (10 ** values['voltage_scale'])):.2f}{inverter.registers['p1_voltage'][6]}")
            print(f"\tPhase 2 voltage: {(values['p2_voltage'] * (10 ** values['voltage_scale'])):.2f}{inverter.registers['p2_voltage'][6]}")
            print(f"\tPhase 3 voltage: {(values['p3_voltage'] * (10 ** values['voltage_scale'])):.2f}{inverter.registers['p3_voltage'][6]}")
            print(f"\tPhase 1-N voltage: {(values['p1n_voltage'] * (10 ** values['voltage_scale'])):.2f}{inverter.registers['p1n_voltage'][6]}")
            print(f"\tPhase 2-N voltage: {(values['p2n_voltage'] * (10 ** values['voltage_scale'])):.2f}{inverter.registers['p2n_voltage'][6]}")
            print(f"\tPhase 3-N voltage: {(values['p3n_voltage'] * (10 ** values['voltage_scale'])):.2f}{inverter.registers['p3n_voltage'][6]}")
        else:
            print(f"\tVoltage: {(values['p1_voltage'] * (10 ** values['voltage_scale'])):.2f}{inverter.registers['p1_voltage'][6]}")

        print(f"\tFrequency: {(values['frequency'] * (10 ** values['frequency_scale'])):.2f}{inverter.registers['frequency'][6]}")
        print(f"\tPower: {(values['power_ac'] * (10 ** values['power_ac_scale'])):.2f}{inverter.registers['power_ac'][6]}")
        print(f"\tPower (Apparent): {(values['power_apparent'] * (10 ** values['power_apparent_scale'])):.2f}{inverter.registers['power_apparent'][6]}")
        print(f"\tPower (Reactive): {(values['power_reactive'] * (10 ** values['power_reactive_scale'])):.2f}{inverter.registers['power_reactive'][6]}")
        print(f"\tPower Factor: {(values['power_factor'] * (10 ** values['power_factor_scale'])):.2f}{inverter.registers['power_factor'][6]}")
        print(f"\tTotal Energy: {(values['energy_total'] * (10 ** values['energy_total_scale']))}{inverter.registers['energy_total'][6]}")

        print(f"\tDC Current: {(values['current_dc'] * (10 ** values['current_dc_scale'])):.2f}{inverter.registers['current_dc'][6]}")
        print(f"\tDC Voltage: {(values['voltage_dc'] * (10 ** values['voltage_dc_scale'])):.2f}{inverter.registers['voltage_dc'][6]}")
        print(f"\tDC Power: {(values['power_dc'] * (10 ** values['power_dc_scale'])):.2f}{inverter.registers['power_dc'][6]}")

nmakel, 2020-12-18:
> Is it possible to run the script below as "server" and then request the information again with the dongle as client?

I'm not quite sure what you're suggesting. When the inverter RS485 port is configured for "Non-Solaredge Logger" it acts as a server, which means a client can get the values like those in the script you posted. When you set the inverter RS485 port to "Modbus multi-device" the inverter will try to connect to other modbus devices as a client. If you want the inverter to get information from a meter it needs to be in "Modbus multi-device" configuration, and will always be a client.

Maikel-K, 2020-12-18:
Okay that's clear, in multi device mode it's a client not a server.

Mmm than I really need to find out WHY it's not working. I need to see if I can make some kind of test setup which will reply what I send to it.

Maybe any tips on that?

nmakel, 2020-12-18:
> Okay that's clear, in multi device mode it's a client not a server.
> 
> Mmm than I really need to find out WHY it's not working. I need to see if I can make some kind of test setup which will reply what I send to it.
> 
> Maybe any tips on that?

Not much I can do to help. Either I need one of those units to experiment with or someone else needs to share their experiences with a different type of gateway.

Maikel-K, 2020-12-18:
Would it have samenhing to do with the following?

1.9.1. UART Free-Frame
The converter support UART Free-Frame function, which will check the intervals between any two bytes when receiving UART data. If the interval time exceed the set value, it will register it as the end of one frame, or it will receive data utill the internal buffer is full, (Default: 512, the largest 1400 bytes), then transfer to Socket Channel.
The default UART Free-Frame interval time is 50ms, it will be packaged into another frame if received UART data interval time is greater than 50ms. The user can also set this interval time to minimum 10ms through Cli command and webpage.
If interval time is set to 10ms and customer MCU can’t send next byte within 10ms, the serial data will be break into two frame.

nmakel, 2020-12-19:
> Would it have samenhing to do with the following?

Maybe? Is it on? Have you tried turning it off?

Maikel-K, 2020-12-19:
Buffer and gap size is always on. Any idea what those should be? Maybe that causes the unwanted data?

[image]

nmakel, 2020-12-19:
You could try setting the buffer size in smaller or larger halves (e.g., 256, 128, or 1024, 2048), and the gap time in 10s of ms (e.g., 40, 30, 60, 70). Run the test script and see if you get better results.

Maikel-K, 2020-12-19:
Will play with that and otherwise I will connect my other SDM unit to the gateway and take the usb-rs485 dongle of that for the inverter. That worked like a charm 😛

Maikel-K, 2020-12-19:
Maybe a bit much to ask but could you hook your rs485 converter that you use for the SDM to the meter to check which settings you need on yours?

nmakel, 2020-12-19:
> Maybe a bit much to ask but could you hook your rs485 converter that you use for the SDM to the meter to check which settings you need on yours?

It's not readily accessible. I may get another tgw715 for the inverter to try it out, but it's not on my to-do for a while.

Maikel-K, 2020-12-19:
I understand, for you it's working okay and I just want it to connect to tcp but that's just for ease of mind that if the rpi reboots I don't have to check which USB port is what again because the usb converter doesn't have any unique info to do persistent names to it.

Anyhow, can you once more tell me how the data should look like when using the test script? Then I can try and start to play around more with those buffersize settings etc.

nmakel, 2020-12-20:
> Anyhow, can you once more tell me how the data should look like when using the test script? Then I can try and start to play around more with those buffersize settings etc.

```
2020-12-14 17:00:28 DEBUG: Getting Frame - 0x3 0x3 0xf2 0x0 0x6
2020-12-14 17:00:28 DEBUG: Factory Request[ReadHoldingRegistersRequest: 3]
2020-12-14 17:00:28 DEBUG: Frame advanced, resetting header!!
2020-12-14 17:00:28 DEBUG: validate: fc-[3] address-1011: count-6
2020-12-14 17:00:28 DEBUG: getValues fc-[3] address-1011: count-6
2020-12-14 17:00:28 DEBUG: send: [ReadRegisterResponse (6)]- b'02030c999a439000000000000000008deb'
```

[20 more replies]

closed 2020-12-30 by nmakel