nmakel/solaredge_meterproxy #7: Difference in exported power

Maikel-K, 2020-12-28:
Hi Niels,

Script is running fine now with RTU, thanks again!

Im seeing difference in what Solaredge is saying I exported and my grid meter. 

I have compared the actual power import/export and those seem to match with the actual power the grid meter is showing.

Any clues what could cause this?

nmakel, 2020-12-28:
Hi Maikel,

Can you be a bit more specific?

- Where are you getting the reading from?
- What are you comparing these values with?
- Do you mean the power (W), or energy (Wh)?
- How big are the differences?

Maikel-K, 2020-12-28:
Hi Niels,

- Solaredge App and Domoticz
- See above
- Wh
- Solaredge 0,55 kWh versus Domoticz 0.103 kWh

Both have been up and running the whole day.

nmakel, 2020-12-28:
Where is domoticz getting its data?

Maikel-K, 2020-12-28:
Through a P1 meter script running via ESP8266 with ESPEasy

Maikel-K, 2020-12-31:
Which data does solaredge use for those actual power and the exported kWh?

I almost got my Mqtt output data running from the SDM630 so I can start logging that in Domoticz too. Then i can compare SDM630 with grid meter and Solaredge too.

nmakel, 2021-01-01:
My guess is `export_energy_active` and `import_energy_active`. Those are the values that correspond to the import/export cumulative values SolarEdge displays for the import and export meters, and with which you can calculate all other values. For cosmetic purposes they probably just show `power_active` in the app.

Maikel-K, 2021-01-01:
I have the idea that the export and own consumption values are mixed. Could this be possible?

Own consumption in Solaredge says 0,78 but grid meter says 0,69 exported.

But I need to fix the Mqtt post to Domoticz with the data of the SDM630

nmakel, 2021-01-01:
> Own consumption in Solaredge says 0,78 but grid meter says 0,69 exported.

These values can both be true.

SolarEdge should also be reporting what it thinks was exported to grid. This value should be similar to what the SDM630 reports as exported active energy.

Use the sdm_modbus tool to check your SDM values. If it reports negative power values at night you need to set the inverted parameter in the configuration file.

Maikel-K, 2021-01-03:
Hi Niels,

Got the SDM630 reporting to Domoticz now so I can start comparing data between the SDM630 and my grid meter.

Question that maybe you can answer: I'm reading the meter in a python script with a while true: loop.
I installed a service file running that script and the loop is set to 10 seconds. But in domoticz I see that the sensor is updated every second.

Any hints?

[Unit] 
Description=EastronSDM630 
After=multi-user.target 

[Service] 
Type=idle 
ExecStart=/usr/bin/python3 test630.py 192.168.1.108 8899 
WorkingDirectory=/home/pi/sdmmodbus 
User=pi 
Restart=always 
RestartSec=60 

[Install]
WantedBy=multi-user.target

Script:

#!/usr/bin/env python3

import argparse
import json
import sdm_modbus
import time
#from datetime import date
import urllib.request

if __name__ == "__main__":
    argparser = argparse.ArgumentParser()
    argparser.add_argument("host", type=str, help="ModbusTCP address")
    argparser.add_argument("port", type=int, help="ModbusTCP port")
    argparser.add_argument("--timeout", type=int, default=1, help="Connection timeout")
    argparser.add_argument("--unit", type=int, default=1, help="Modbus unit")
    argparser.add_argument("--json", action="store_true", default=False, help="Output as JSON")
    args = argparser.parse_args()

    meter = sdm_modbus.SDM630(
        host=args.host,
        port=args.port,
        timeout=args.timeout,
        unit=args.unit
    )

while True:
 ActivePower = (meter.read("total_power_active"))
 ACTIVEPOW = "{0:0.0f}".format(ActivePower)
 ImportActiveEnergy = (meter.read("import_energy_active")) 
 IMPORT = float (1000) * float ("{0:0.3f}".format(ImportActiveEnergy)) 
 ExportActiveEnergy = (meter.read("export_energy_active"))
 EXPORT = float (1000) * float ("{0:0.3f}".format(ExportActiveEnergy))

 if int(ACTIVEPOW) < 0:
   PRODUCTION = int(ACTIVEPOW) * -1
   CONSUMPTION = 0
 else:
   PRODUCTION = 0
   CONSUMPTION = int(ACTIVEPOW)

# --------------------------------------------------
# END READING RS485-INTERFACE
# --------------------------------------------------
# START OF UPLOAD TO DOMOTICZ
# --------------------------------------------------

# Presetting of Domoticz server, port and IDX for YOUR virtual sensors
 domoticz="http://192.168.1.104" # Change to the ip-address of Domoticz
 domoticzport = "8080"  		# Change to your port
 domoticz_P1 = "3185"    		# IDX for virtual P1-meter

# Setting & Linking of parameters
 USAGE1 = IMPORT
 USAGE2 = 0
 RETURN1 = EXPORT
 RETURN2 = 0
 CONS = CONSUMPTION
 PROD = PRODUCTION

#print USAGE1
#print USAGE2
#print RETURN1
#print RETURN2
#print CONS
#print PROD

 httpresponse = urllib.request.urlopen("" + domoticz + ":" + domoticzport + "/json.htm?type=command&param=udevice&idx=" + str(domoticz_P1) + "&nvalue=0&svalue=" + str(float(USAGE1)) + ";" + str(float(USAGE2)) + ";" + str(float(RETURN1)) + ";" + str(float(RETURN2)) + ";" + str(float(CONS)) + ";" + str(float(PROD)) )

 #print ('Completed succesfully and script finished')
time.sleep(10.0)

nmakel, 2021-01-03:
I don't use Domoticz, and am not familiar with the ways they record, present and maybe interpolate data. Check the sample rates of any graphs or visualizations, perhaps that is the problem. Or the script is crashing and restarted by systemd every second. Who knows.

Closing this issue as it is not related to solaredge_meterproxy.

closed 2021-01-03 by nmakel