nmakel/solaredge_meterproxy #13: SDM630 and missing values

SebuZet, 2022-01-04:
Hi,

May I ask why values like "l1_energy_active", "l2_energy_active", "l3_energy_active", "l1_import_energy_active", "l2_import_energy_active", "l3_import_energy_active", "l1_export_energy_active", "l2_export_energy_active", "l3_export_energy_active", "l1_energy_reactive", "l2_energy_reactive", "l3_energy_reactive", "l1_energy_apparent", "l2_energy_apparent", "l3_energy_apparent" are not taken from meter - it looks like those values are available in data from this meter?
Is there some cose why it is not passed to SE?

Regards,
Sebu

nmakel, 2022-01-05:
Can you confirm the registers contain data?

SebuZet, 2022-01-05:
Yes, I can confirm that import data looks good. I do not have PV yet (nor SolarEdge inverter) so I don't export any energy.
I've SDM630 v2 meter.
I've extended ESPHome sdmxxx meter component locally to get those data. I want to port your meter proxy to ESPHome component and I was wondering if you skip those data by purpose.

nmakel, 2022-01-06:
There should be no downsided to passing the values along in that case. Good luck with your project!