nmakel/solaredge_meterproxy #15: Question: feed values from API to inverter?

tobiasehlert, 2022-01-14:
Nice project @nmakel!

I got two questions:

1. Is it possible to add an "api-device", where you can point to a data source providing measurements similar to the devices you support. And with this I am thinking to parse my providers load-monitor ([Tibber Pulse](https://tibber.com/se/store/produkt/pulse) (Swedish page)) and return that data back to this application in a giant JSON file or so. This could of course also be adapted to other providers, if there is some data available, as long as it's correct formatted.
What you think?

2. Is it possible to both read modbus over tcp from the inverter (to collect metrics) and feed the inverter with information over tcp (add measurements at the mainfuses) at the same time?

Kind regards,
Tobias

nmakel, 2022-01-14:
> Nice project @nmakel!

Thanks!
> 1. Is it possible to add an "api-device", where you can point to a data source providing measurements similar to the devices you support. And with this I am thinking to parse my providers [...]

Sure, a script which uses the requests library to grab a JSON object from an API endpoint should be easy to implement. I recommend you have a look at the already implemented devices to get a feel for what needs to be collected.

>     2. Is it possible to both read modbus over tcp from the inverter (to collect metrics) and feed the inverter with information over tcp (add measurements at the mainfuses) at the same time?

Absolutely, I have a setup as you describe. I use [solaredge_modbus](https://github.com/nmakel/solaredge_modbus) over TCP to collect inverter data, and run `solaredge_meterproxy` to feed data back via the RS485 port. It should also be possible to use `solaredge_modbus` to collect data via RS485 port 1, and send data back using `solaredge_meterproxy` on RS485 port 2. You will need to configure both ports appropriately on the inverter.

tobiasehlert, 2022-01-14:
> Sure, a script which uses the requests library to grab a JSON object from an API endpoint should be easy to implement. I recommend you have a look at the already implemented devices to get a feel for what needs to be collected.

Alright awesome! I could write something in go to be able to generate a JSON with those values you have in the other devices.
Would it be easy for you to add an api-device in Python then? I've never written Python 😄 

> I'm afraid not. The inverter only communicates with meters over the RS485 bus.

Damnit!
But would it go to provide data over RS485 to the inverter but connect to it from another device collecting all metrics over TCP?

nmakel, 2022-01-14:
> Alright awesome! I could write something in go to be able to generate a JSON with those values you have in the other devices. Would it be easy for you to add an api-device in Python then? I've never written Python smile

It feels like you're overcomplicating things. Given an API you should be able to retrieve its data using a library such as [requests](https://docs.python-requests.org/en/latest/). This yields JSON/XML/CSV which should be simple to insert into the dictionary as documented in [generic.py](https://github.com/nmakel/solaredge_meterproxy/blob/master/devices/generic.py).

This method should be trivial to adapt to any existing API or device which speaks HTTP and returns JSON/XML/CSV.

> Damnit! But would it go to provide data over RS485 to the inverter but connect to it from another device collecting all metrics over TCP?

I'm currently collecting data from my inverter using [solaredge_modbus](https://github.com/nmakel/solaredge_modbus) via Modbus TCP. At the same time I am providing meter values to the inverter over RS485 using this `solaredge_meterproxy` script. The only limitation is that the inverter's Modbus TCP connection only allows a single connection at a time.

tobiasehlert, 2022-01-14:
> It feels like you're overcomplicating things. Given an API you should be able to retrieve its data using a library such as [requests](https://docs.python-requests.org/en/latest/). This yields [...]

There seems to be a Python library for Tibber ([on GitHub](https://github.com/Danielhiversen/pyTibber)) but to get realtime consumption, it's needed to have a graphql subscription. I don't know how to integrate that fully.. but you might want to look?

There is an [API explorer](https://developer.tibber.com/explorer) on their developer page, where you can get a demo token and then paste following JSON:
```graphql
subscription {
  liveMeasurement(homeId: "cc83e83e-8cbf-4595-9bf7-c3cf192f7d9c") {
    timestamp
    power
    powerProduction
    powerFactor
    voltagePhase1
    voltagePhase2
    voltagePhase3
    currentL1
    currentL2
    currentL3
  }
}
```

The return looks something like this (and gets updated very frequently):
```json
{
  "liveMeasurement": {
    "timestamp": "2022-01-14T17:19:40.000+01:00",
    "power": 2207,
    "powerProduction": 0,
    "powerFactor": 0.979,
    "voltagePhase1": 218,
    "voltagePhase2": 221,
    "voltagePhase3": 221,
    "currentL1": 1.98,
    "currentL2": 5.87,
    "currentL3": 3.02
  }
}
```

Or is there a major lack of values that would need to be provided? 🤔

nmakel, 2022-03-03:
> > It feels like you're overcomplicating things. Given an API you should be able to retrieve its data using a library such as [requests](https://docs.python-requests.org/en/latest/). This yields JSON/XML/CSV which should be simple to insert into the dictionary as documented in [generic.py](https://github.com/nmakel/solaredge_meterproxy/blob/master/devices/generic.py).
> > This method should be trivial to adapt to any existing API or device which speaks HTTP and returns JSON/XML/CSV.
> 
> There seems to be a Python library for Tibber ([on GitHub](https://github.com/Danielhiversen/pyTibber)) but to get realtime consumption, it's needed to have a graphql subscription. I don't know how to integrate that fully.. but you might want to look?
> 
> There is an [API explorer](https://developer.tibber.com/explorer) on their developer page, where you can get a demo token and then paste following JSON:
> 
> ```graphql
> subscription {
>   liveMeasurement(homeId: "cc83e83e-8cbf-4595-9bf7-c3cf192f7d9c") {
>     timestamp
>     power
>     powerProduction
>     powerFactor
>     voltagePhase1
>     voltagePhase2
>     voltagePhase3
>     currentL1
>     currentL2
>     currentL3
>   }
> }
> ```
> 
> The return looks something like this (and gets updated very frequently):
> 
> ```json
> {
>   "liveMeasurement": {
>     "timestamp": "2022-01-14T17:19:40.000+01:00",
>     "power": 2207,
>     "powerProduction": 0,
>     "powerFactor": 0.979,
>     "voltagePhase1": 218,
>     "voltagePhase2": 221,
>     "voltagePhase3": 221,
>     "currentL1": 1.98,
>     "currentL2": 5.87,
>     "currentL3": 3.02
>   }
> }
> ```
> 
> Or is there a major lack of values that would need to be provided? thinking

It isn't obvious which values are absolutely required by the inverter to do what it needs to do. The voltage and current will also give you the power values, and I suspect you will need to get the cumulative import and export kWh from somewhere. The Tibber API is heavily focussed on pricing, which makes sense, but light on actual grid and quality data.