nmakel/solaredge_meterproxy #16: "DEBUG: Frame check failed, ignoring!!" on SE6K and SDM630 V2

SebuZet, 2022-02-24:
Hi,
Im trying to run tool but I've got error message likethis:

```
2022-02-24 07:11:38 DEBUG: setValues[3] 1101:82
2022-02-24 07:11:38 DEBUG: Frame check failed, ignoring!!
2022-02-24 07:11:38 DEBUG: Resetting frame - Current Frame in buffer - 0x80 0x0 0xf8 0x0 0x80 0x0 0x0 0xf8 0x0 0xf8 0x78 0x0 0x0 0x0 0xf8 0x78 0xf 0x80 0x0 0xf8 0x0 0x80 0x0 0x0 0xf8 0x0 0xf8 0x78 0x0 0x0 0x0 0xf8 0x78 0xf
```
My config (kemp-rtu.conf):
```
[server]
device = /dev/ttySC0
baud = 38400
parity = E
#timeout = 1
log_level = INFO
meters = sdm

[sdm]
baud = 38400
device = /dev/ttySC1
dst_address = 2
src_address = 5
type = sdm630
serial_number = 123321
ct_current = 5
ct_inverted = 0
phase_offset = 120
refresh_rate = 5
```

I'm using semp-rtu.py

I can confirm that connection with SDM meter is working correctly.
What am I doing wrong?

SebuZet, 2022-02-28:
Can anyone help mee with this? I've tried to set framer = rtu in [server] section but it didn't help.
It look like problems with SE connection

SebuZet, 2022-02-28:
Ok,
I've managed connection problems. It looks like my SE doesn't change baud rate as expected. Now it is (almost) working.
SE shows that meter status is OK.
When I run exmaple.py (from solaredge_modbus module) I've got such output:

```
{
    "c_manufacturer": "SolarEdge",
    "c_model": "SE6K-RW0TEBNN4",
    "c_version": "0004.0014.0228",
    "c_serialnumber": "XX",
    "c_deviceaddress": 1,
    "c_sunspec_did": 103,
    "current": 0,
    "l1_current": 0,
    "l2_current": 0,
    "l3_current": 0,
    "current_scale": -2,
    "l1_voltage": 4150,
    "l2_voltage": 4179,
    "l3_voltage": 4162,
    "l1n_voltage": 2396,
    "l2n_voltage": 2413,
    "l3n_voltage": 2402,
    "voltage_scale": -1,
    "power_ac": 0,
    "power_ac_scale": 0,
    "frequency": 5001,
    "frequency_scale": -2,
    "power_apparent": 0,
    "power_apparent_scale": 0,
    "power_reactive": 0,
    "power_reactive_scale": 0,
    "power_factor": 0,
    "power_factor_scale": 0,
    "energy_total": 0,
    "energy_total_scale": 0,
    "current_dc": 0,
    "current_dc_scale": 0,
    "voltage_dc": 3,
    "voltage_dc_scale": -1,
    "power_dc": 0,
    "power_dc_scale": 0,
    "temperature": 2040,
    "temperature_scale": -2,
    "status": 2,
    "vendor_status": 0,
    "rrcr_state": 0,
    "active_power_limit": 100,
    "cosphi": 0,
    "meters": {
        "Meter1": {
            "c_manufacturer": "SolarEdge",
            "c_model": "SE-RGMTR-1D-240C-A",
            "c_option": "Export+Import",
            "c_version": "0",
            "c_serialnumber": "0",
            "c_deviceaddress": 2,
            "c_sunspec_did": 201,
            "current": -32768,
            "l1_current": -32768,
            "l2_current": -32768,
            "l3_current": -32768,
            "current_scale": -32768,
            "voltage_ln": -32768,
            "l1n_voltage": -32768,
            "l2n_voltage": -32768,
            "l3n_voltage": -32768,
            "voltage_ll": -32768,
            "l12_voltage": -32768,
            "l23_voltage": -32768,
            "l31_voltage": -32768,
            "voltage_scale": -32768,
            "frequency": -32768,
            "frequency_scale": -32768,
            "power": -32768,
            "l1_power": -32768,
            "l2_power": -32768,
            "l3_power": -32768,
            "power_scale": -32768,
            "power_apparent": -32768,
            "l1_power_apparent": -32768,
            "l2_power_apparent": -32768,
            "l3_power_apparent": -32768,
            "power_apparent_scale": -32768,
            "power_reactive": -32768,
            "l1_power_reactive": -32768,
            "l2_power_reactive": -32768,
            "l3_power_reactive": -32768,
            "power_reactive_scale": -32768,
            "power_factor": -32768,
            "l1_power_factor": -32768,
            "l2_power_factor": -32768,
            "l3_power_factor": -32768,
            "power_factor_scale": -32768,
            "export_energy_active": 0,
            "l1_export_energy_active": 0,
            "l2_export_energy_active": 0,
            "l3_export_energy_active": 0,
            "import_energy_active": 0,
            "l1_import_energy_active": 0,
            "l2_import_energy_active": 0,
            "l3_import_energy_active": 0,
            "energy_active_scale": 0,
            "export_energy_apparent": 0,
            "l1_export_energy_apparent": 0,
            "l2_export_energy_apparent": 0,
            "l3_export_energy_apparent": 0,
            "import_energy_apparent": 0,
            "l1_import_energy_apparent": 0,
            "l2_import_energy_apparent": 0,
            "l3_import_energy_apparent": 0,
            "energy_apparent_scale": -32768,
            "import_energy_reactive_q1": 0,
            "l1_import_energy_reactive_q1": 0,
            "l2_import_energy_reactive_q1": 0,
            "l3_import_energy_reactive_q1": 0,
            "import_energy_reactive_q2": 0,
            "l1_import_energy_reactive_q2": 0,
            "l2_import_energy_reactive_q2": 0,
            "l3_import_energy_reactive_q2": 0,
            "export_energy_reactive_q3": 0,
            "l1_export_energy_reactive_q3": 0,
            "l2_export_energy_reactive_q3": 0,
            "l3_export_energy_reactive_q3": 0,
            "export_energy_reactive_q4": 0,
            "l1_export_energy_reactive_q4": 0,
            "l2_export_energy_reactive_q4": 0,
            "l3_export_energy_reactive_q4": 0,
            "energy_reactive_scale": -32768
        }
    },
    "batteries": {}
}
```
Data from meter look very strange. What should check?

nmakel, 2022-03-03:
> Ok, I've managed connection problems. It looks like my SE doesn't change baud rate as expected.

Correct. When the SE is a master polling meters it will switch back to 9600 baud, irrespective of the baudrate you set when the SE is in Sunspec mode.

> Now it is (almost) working. SE shows that meter status is OK. When I run exmaple.py (from solaredge_modbus module) I've got such output:

It would be useful to have the `debug` output of semp-rtu.py when the SE is trying to poll your meter. You can enable this much more verbose output by adding  `log_level = DEBUG` to your `[server]` configuration. Please post a few seconds of this output so I can have a look at the register that the SE is asking for.

Also, could you note the settings with which you have configured the meter on the SE?

SebuZet, 2022-03-03:
Thank you for your response.

I've attached log file.
Same part of logs are below:

`
2022-02-28 17:16:27 DEBUG: setValues[3] 1601:23
2022-02-28 17:16:27 DEBUG: [0, 2, 4, 0, 0, 5]
2022-02-28 17:16:27 DEBUG: setValues[3] 1651:6
2022-02-28 17:16:27 DEBUG: [57785, 1, 0, 0, 0, 0, 202, 31, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-02-28 17:16:27 DEBUG: setValues[3] 1701:23
2022-02-28 17:16:27 INFO: Created <Thread(t_update_2, initial)>: sdm sdm630 SDM630(/dev/ttySC1, connectionType.RTU: stopbits=1, parity=N, baud=38400, timeout=1, retries=3, unit=0x5)
2022-02-28 17:16:28 DEBUG: device: SDM630(/dev/ttySC1, connectionType.RTU: stopbits=1, parity=N, baud=38400, timeout=1, retries=3, unit=0x5)
2022-02-28 17:16:28 INFO: Starting <Thread(t_update_2, started 3053687872)>
2022-02-28 17:16:28 DEBUG: Client Connected [/dev/ttySC0:/dev/ttySC0]
2022-02-28 17:16:28 DEBUG: Started thread to serve client
2022-02-28 17:16:28 DEBUG: Current transaction state - IDLE
2022-02-28 17:16:28 DEBUG: Running transaction 1
2022-02-28 17:16:28 DEBUG: SEND: 0x5 0x4 0x0 0x0 0x0 0x50 0xf1 0xb2
2022-02-28 17:16:28 DEBUG: New Transaction state 'SENDING'
2022-02-28 17:16:28 DEBUG: Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-02-28 17:16:28 DEBUG: Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-02-28 17:16:28 DEBUG: RECV: 0x5 0x4 0xa0 0x43 0x6f 0x30 0xe8 0x43 0x6f 0xc2 0x3a 0x43 0x6f 0x3f 0x73 0x3f 0xd5 0x54 0x6f 0x3f 0x1 0x39 0x2d 0x3f 0xef 0x6f 0xd7 0x43 0x65 0xf6 0x56 0x42 0x8e 0x68 0x69 0x43 0x8e 0x33 0xf7 0x43 0xc7 0x52 0x94 0x42 0xf2 0xc 0xd7 0x43 0xdf 0xc4 0xc1 0xc3 0x94 0x9f 0x2c 0xc2 0x93 0x4e 0x2a 0xc3 0x8e 0x39 0x82 0x3f 0x13 0xad 0x11 0x3f 0x16 0x9d 0x78 0x3f 0x22 0xaf 0x8f 0xc2 0x5b 0x18 0xa6 0xc2 0x57 0xdb 0xcf 0xc2 0x4a 0x53 0x90 0x43 0x6f 0x62 0xf3 0x0 0x0 0x0 0x0 0x3f 0xad 0x7 0x7 0x40 0x81 0xc5 0x45 0x0 0x0 0x0 0x0 0x44 0x12 0xf3 0xcd 0x0 0x0 0x0 0x0 0x44 0x72 0x96 0x9c 0x0 0x0 0x0 0x0 0xc4 0x24 0x19 0xb0 0x3f 0x1b 0x13 0xa4 0x0 0x0 0x0 0x0 0xc2 0x52 0xe0 0xff 0x0 0x0 0x0 0x0 0x42 0x47 0x8b 0x5b 0x44 0xa3 0x76 0x14 0x0 0x0 0x0 0x0 0x3c 0xe5 0x60 0x41 0x44 0x91 0x6d 0x91 0x8f 0x57
2022-02-28 17:16:28 DEBUG: Getting Frame - 0x4 0xa0 0x43 0x6f 0x30 0xe8 0x43 0x6f 0xc2 0x3a 0x43 0x6f 0x3f 0x73 0x3f 0xd5 0x54 0x6f 0x3f 0x1 0x39 0x2d 0x3f 0xef 0x6f 0xd7 0x43 0x65 0xf6 0x56 0x42 0x8e 0x68 0x69 0x43 0x8e 0x33 0xf7 0x43 0xc7 0x52 0x94 0x42 0xf2 0xc 0xd7 0x43 0xdf 0xc4 0xc1 0xc3 0x94 0x9f 0x2c 0xc2 0x93 0x4e 0x2a 0xc3 0x8e 0x39 0x82 0x3f 0x13 0xad 0x11 0x3f 0x16 0x9d 0x78 0x3f 0x22 0xaf 0x8f 0xc2 0x5b 0x18 0xa6 0xc2 0x57 0xdb 0xcf 0xc2 0x4a 0x53 0x90 0x43 0x6f 0x62 0xf3 0x0 0x0 0x0 0x0 0x3f 0xad 0x7 0x7 0x40 0x81 0xc5 0x45 0x0 0x0 0x0 0x0 0x44 0x12 0xf3 0xcd 0x0 0x0 0x0 0x0 0x44 0x72 0x96 0x9c 0x0 0x0 0x0 0x0 0xc4 0x24 0x19 0xb0 0x3f 0x1b 0x13 0xa4 0x0 0x0 0x0 0x0 0xc2 0x52 0xe0 0xff 0x0 0x0 0x0 0x0 0x42 0x47 0x8b 0x5b 0x44 0xa3 0x76 0x14 0x0 0x0 0x0 0x0 0x3c 0xe5 0x60 0x41 0x44 0x91 0x6d 0x91
2022-02-28 17:16:28 DEBUG: Factory Response[ReadInputRegistersResponse: 4]
2022-02-28 17:16:28 DEBUG: Frame advanced, resetting header!!
2022-02-28 17:16:28 DEBUG: Adding transaction 5
2022-02-28 17:16:28 DEBUG: Getting transaction 5
2022-02-28 17:16:28 DEBUG: Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2022-02-28 17:16:28 DEBUG: [17263, 12520, 17263, 49722, 17263, 16243, 16341, 21615, 16129, 14637, 16367, 28631, 17253, 63062, 17038, 26729, 17294, 13303, 17351, 21140, 17138, 3287, 17375, 50369, 50068, 40748, 49811, 20010, 50062, 14722, 16147, 44305, 16150, 40312, 16162, 44943, 49755, 6310, 49751, 56271, 49738, 21392, 17263, 25331, 0, 0, 16301, 1799, 16513, 50501, 0, 0, 17426, 62413, 0, 0, 17522, 38556, 0, 0, 50212, 6576, 16155, 5028, 0, 0, 49746, 57599, 0, 0, 16967, 35675, 17571, 30228, 0, 0, 15589, 24641, 17553, 28049]
2022-02-28 17:16:28 DEBUG: [b'Co', b'0\xe8']
2022-02-28 17:16:28 DEBUG: [b'Co', b'\xc2:']
2022-02-28 17:16:28 DEBUG: [b'Co', b'?s']
2022-02-28 17:16:28 DEBUG: [b'?\xd5', b'To']
2022-02-28 17:16:28 DEBUG: [b'?\x01', b'9-']
2022-02-28 17:16:28 DEBUG: [b'?\xef', b'o\xd7']
2022-02-28 17:16:28 DEBUG: [b'Ce', b'\xf6V']
2022-02-28 17:16:28 DEBUG: [b'B\x8e', b'hi']
2022-02-28 17:16:28 DEBUG: [b'C\x8e', b'3\xf7']
2022-02-28 17:16:28 DEBUG: [b'C\xc7', b'R\x94']
2022-02-28 17:16:28 DEBUG: [b'B\xf2', b'\x0c\xd7']
2022-02-28 17:16:28 DEBUG: [b'C\xdf', b'\xc4\xc1']
2022-02-28 17:16:28 DEBUG: [b'\xc3\x94', b'\x9f,']
2022-02-28 17:16:28 DEBUG: [b'\xc2\x93', b'N*']
2022-02-28 17:16:28 DEBUG: [b'\xc3\x8e', b'9\x82']
2022-02-28 17:16:28 DEBUG: [b'?\x13', b'\xad\x11']
2022-02-28 17:16:28 DEBUG: [b'?\x16', b'\x9dx']
2022-02-28 17:16:28 DEBUG: [b'?"', b'\xaf\x8f']
2022-02-28 17:16:28 DEBUG: [b'\xc2[', b'\x18\xa6']
2022-02-28 17:16:28 DEBUG: [b'\xc2W', b'\xdb\xcf']
2022-02-28 17:16:28 DEBUG: [b'\xc2J', b'S\x90']
2022-02-28 17:16:28 DEBUG: [b'Co', b'b\xf3']
2022-02-28 17:16:28 DEBUG: [b'?\xad', b'\x07\x07']
2022-02-28 17:16:28 DEBUG: [b'@\x81', b'\xc5E']
2022-02-28 17:16:28 DEBUG: [b'D\x12', b'\xf3\xcd']
2022-02-28 17:16:28 DEBUG: [b'Dr', b'\x96\x9c']
2022-02-28 17:16:28 DEBUG: [b'\xc4$', b'\x19\xb0']
2022-02-28 17:16:28 DEBUG: [b'?\x1b', b'\x13\xa4']
2022-02-28 17:16:28 DEBUG: [b'\xc2R', b'\xe0\xff']
2022-02-28 17:16:28 DEBUG: [b'BG', b'\x8b[']
2022-02-28 17:16:28 DEBUG: [b'D\xa3', b'v\x14']
2022-02-28 17:16:28 DEBUG: [b'\x00\x00', b'\x00\x00']
2022-02-28 17:16:28 DEBUG: [b'<\xe5', b'`A']
2022-02-28 17:16:28 DEBUG: [b'D\x91', b'm\x91']
[... cut ...] x0 0x0 0x0 0x0 0x42 0x47 0x9e 0x6 0x44 0xa3 0x76 0x14 0x0 0x0 0x0 0x0 0x3c 0xe5 0x60 0x41 0x44 0x91 0x6d 0x99
2022-02-28 17:16:30 DEBUG: Factory Response[ReadInputRegistersResponse: 4]
2022-02-28 17:16:30 DEBUG: Frame advanced, resetting header!!
`
I've set up meter as WattNode. I was following up your instructions.
SE addr: 1
Meter Addr: 2
I can provide screenshots from setApp application. Meter set as I+E.

[log.txt.zip](https://github.com/nmakel/solaredge_meterproxy/files/8177463/log.txt.zip)

SebuZet, 2022-03-03:
I think that this output is from try with your static data (from another issue).

`
2022-02-28 21:31:48 DEBUG: [0, 0, 5, 5, 5, 5, 0, 1, 0, 15, 1, 10000, 10000, 10000, 64536, 64536, 64536, 1500, 120, 0, 0, 20000, 0]
2022-02-28 21:31:48 DEBUG: setValues[3] 1601:23
2022-02-28 21:31:48 DEBUG: [0, 2, 4, 0, 0, 5]
2022-02-28 21:31:48 DEBUG: setValues[3] 1651:6
2022-02-28 21:31:48 DEBUG: [57785, 1, 0, 0, 0, 0, 202, 31, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2022-02-28 21:31:48 DEBUG: setValues[3] 1701:23
2022-02-28 21:31:48 INFO: Created <Thread(t_update_2, initial)>: sdm setdata True
2022-02-28 21:31:49 INFO: return set data set from NMakel
2022-02-28 21:31:49 INFO: Starting <Thread(t_update_2, started 3053962304)>
2022-02-28 21:31:49 DEBUG: [27441, 17880, 22036, 17761, 27441, 17880, 22036, 17761, 62259, 50059, 0, 0, 0, 0, 0, 0, 58982, 17260, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 52429, 16967]
2022-02-28 21:31:49 DEBUG: setValues[3] 1001:34
2022-02-28 21:31:49 DEBUG: Client Connected [/dev/ttySC0:/dev/ttySC0]
2022-02-28 21:31:49 DEBUG: Started thread to serve client
2022-02-28 21:31:49 DEBUG: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32846, 17743, 32846, 17743, 0, 0, 0, 0, 0, 0, 24946, 17941, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13705, 48852, 0, 0, 0, 0, 0, 0, 42598, 50201, 0, 0, 0, 0, 0, 0, 56372, 17448, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 26152, 17480, 0, 0, 13366, 17745, 0, 0, 0, 0, 0, 0, 0, 0]
2022-02-28 21:31:49 DEBUG: setValues[3] 1101:82
2022-02-28 21:31:50 DEBUG: Getting Frame - 0x3 0x6 0x40 0x0 0x17
2022-02-28 21:31:50 DEBUG: Factory Request[ReadHoldingRegistersRequest: 3]
2022-02-28 21:31:50 DEBUG: Frame advanced, resetting header!!
2022-02-28 21:31:50 DEBUG: validate: fc-[3] address-1601: count-23
2022-02-28 21:31:50 DEBUG: getValues fc-[3] address-1601: count-23
2022-02-28 21:31:50 DEBUG: send: [ReadHoldingRegistersResponse (23)]- b'02032e000000000005000500050005000000010000000f0001271027102710fc18fc18fc1805dc0078000000004e2000000a36'
2022-02-28 21:31:51 INFO: return set data set from NMakel
2022-02-28 21:31:51 DEBUG: [27441, 17880, 22036, 17761, 27441, 17880, 22036, 17761, 62259, 50059, 0, 0, 0, 0, 0, 0, 58982, 17260, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 52429, 16967]
2022-02-28 21:31:51 DEBUG: Getting Frame - 0x3 0x6 0x40 0x0 0x17
2022-02-28 21:31:51 DEBUG: Factory Request[ReadHoldingRegistersRequest: 3]
2022-02-28 21:31:51 DEBUG: Frame advanced, resetting header!!
2022-02-28 21:31:51 DEBUG: validate: fc-[3] address-1601: count-23
2022-02-28 21:31:51 DEBUG: getValues fc-[3] address-1601: count-23
2022-02-28 21:31:51 DEBUG: send: [ReadHoldingRegistersResponse (23)]- b'02032e000000000005000500050005000000010000000f0001271027102710fc18fc18fc1805dc0078000000004e2000000a36'
2022-02-28 21:31:51 DEBUG: setValues[3] 1001:34
2022-02-28 21:31:51 DEBUG: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32846, 17743, 32846, 17743, 0, 0, 0, 0, 0, 0, 24946, 17941, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13705, 48852, 0, 0, 0, 0, 0, 0, 42598, 50201, 0, 0, 0, 0, 0, 0, 56372, 17448, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 26152, 17480, 0, 0, 13366, 17745, 0, 0, 0, 0, 0, 0, 0, 0]
2022-02-28 21:31:51 DEBUG: setValues[3] 1101:82
2022-02-28 21:31:52 DEBUG: Getting Frame - 0x6 0x6 0x47 0x0 0x0
2022-02-28 21:31:52 DEBUG: Factory Request[WriteSingleRegisterRequest: 6]
2022-02-28 21:31:52 DEBUG: Frame advanced, resetting header!!
2022-02-28 21:31:52 DEBUG: validate: fc-[6] address-1608: count-1
2022-02-28 21:31:52 DEBUG: setValues[6] 1608:1
2022-02-28 21:31:52 DEBUG: getValues fc-[6] address-1608: count-1
2022-02-28 21:31:52 DEBUG: send: [WriteRegisterResponse 1607 => 0]- b'0206064700003964'
2022-02-28 21:31:53 INFO: return set data set from NMakel
2022-02-28 21:31:53 DEBUG: [27441, 17880, 22036, 17761, 27441, 17880, 22036, 17761, 62259, 50059, 0, 0, 0, 0, 0, 0, 58982, 17260, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 52429, 16967]
2022-02-28 21:31:53 DEBUG: setValues[3] 1001:34
2022-02-28 21:31:53 DEBUG: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32846, 17743, 32846, 17743, 0, 0, 0, 0, 0, 0, 24946, 17941, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13705, 48852, 0, 0, 0, 0, 0, 0, 42598, 50201, 0, 0, 0, 0, 0, 0, 56372, 17448, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 26152, 17480, 0, 0, 13366, 17745, 0, 0, 0, 0, 0, 0, 0, 0]
2022-02-28 21:31:53 DEBUG: setValues[3] 1101:82
2022-02-28 21:31:53 DEBUG: Getting Frame - 0x3 0x6 0x40 0x0 0x17
2022-02-28 21:31:53 DEBUG: Factory Request[ReadHoldingRegistersRequest: 3]
2022-02-28 21:31:53 DEBUG: Frame advanced, resetting header!!
2022-02-28 21:31:53 DEBUG: validate: fc-[3] address-1601: count-23
2022-02-28 21:31:53 DEBUG: getValues fc-[3] address-1601: count-23
2022-02-28 21:31:53 DEBUG: send: [ReadHoldingRegistersResponse (23)]- b'02032e000000000005000500050005000000000000000f0001271027102710fc18fc18fc1805dc0078000000004e2000009bf6'
2022-02-28 21:31:54 DEBUG: Getting Frame - 0x6 0x6 0x47 0x0 0x0
2022-02-28 21:31:54 DEBUG: Factory Request[WriteSingleRegisterRequest: 6]
2022-02-28 21:31:54 DEBUG: Frame advanced, resetting header!!
2022-02-28 21:31:54 DEBUG: validate: fc-[6] address-1608: count-1
2022-02-28 21:31:54 DEBUG: setValues[6] 1608:1
2022-02-28 21:31:54 DEBUG: getValues fc-[6] address-1608: count-1
2022-02-28 21:31:54 DEBUG: send: [WriteRegisterResponse 1607 => 0]- b'0206064700003964'
2022-02-28 21:31:55 INFO: return set data set from NMakel
[... cut ...]
2022-02-28 21:31:55 DEBUG: setValues[3] 1101:82
2022-02-28 21:31:56 DEBUG: Getting Frame - 0x6 0x6 0x47 0x0 0x0
2022-02-28 21:31:56 DEBUG: Factory Request[WriteSingleRegisterRequest: 6]
2022-02-28 21:31:56 DEBUG: Frame advanced, resetting header!!
2022-02-28 21:31:56 DEBUG: validate: fc-[6] address-1608: count-1
2022-02-28 21:31:56 DEBUG: setValues[6] 1608:1
2022-02-28 21:31:56 DEBUG: getValues fc-[6] address-1608: count-1
2022-02-28 21:31:56 DEBUG: send: [WriteRegisterResponse 1607 => 0]- b'0206064700003964'
`

nmakel, 2022-03-03:
> 

So a quick glance tells me two things: (1) the SDM630 is responding as expected and is providing sensible readings, and (2) the SE is communicating with `semp-rtu.py` and repeatedly asking for block 1601 values which contain basic configuration values of the meter.

It would be helpful to know whether the SE asks for blocks 1651 and 1701 at any point, most probably after (re)booting. If so, this means the SE is able to receive and decode the responses.

When the SE is satisfied we would expect it to start polling 1001 and 1101 multiple times per second. Looks like the SE isn't happy, and never gets to that stage. This would suggest something is missing, or incorrect in blocks 1601, 1651 or 1701.

SebuZet, 2022-03-04:
Thank you for replay.
Yes, SE is asking for 1701 register.

This is list of register SE is asking for:

```
pi@raspberrypi:~/solaredge_meterproxy $ cat out3.txt | grep setValues
2022-03-04 08:15:53 DEBUG: setValues[3] 1601:23
2022-03-04 08:15:53 DEBUG: setValues[3] 1651:6
2022-03-04 08:15:53 DEBUG: setValues[3] 1701:23
2022-03-04 08:15:54 DEBUG: setValues[3] 1001:34
2022-03-04 08:15:54 DEBUG: setValues[3] 1101:82
2022-03-04 08:15:56 DEBUG: setValues[3] 1001:34
2022-03-04 08:15:56 DEBUG: setValues[3] 1101:82
2022-03-04 08:15:58 DEBUG: setValues[3] 1001:34
2022-03-04 08:15:58 DEBUG: setValues[3] 1101:82
2022-03-04 08:15:59 DEBUG: setValues[6] 1608:1
2022-03-04 08:16:01 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:01 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:02 DEBUG: setValues[6] 1608:1
2022-03-04 08:16:03 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:03 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:05 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:05 DEBUG: setValues[6] 1608:1
2022-03-04 08:16:05 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:07 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:07 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:09 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:09 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:11 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:11 DEBUG: setValues[6] 1608:1
2022-03-04 08:16:11 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:13 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:13 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:15 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:15 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:17 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:17 DEBUG: setValues[6] 1608:1
2022-03-04 08:16:17 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:19 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:19 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:21 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:21 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:23 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:23 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:25 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:25 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:27 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:27 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:28 DEBUG: setValues[6] 1608:1
2022-03-04 08:16:29 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:29 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:30 DEBUG: setValues[6] 1603:1
2022-03-04 08:16:31 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:31 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:33 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:33 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:35 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:35 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:36 DEBUG: setValues[6] 1603:1
2022-03-04 08:16:37 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:37 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:39 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:39 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:40 DEBUG: setValues[6] 1603:1
2022-03-04 08:16:41 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:41 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:43 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:43 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:44 DEBUG: setValues[6] 1603:1
[... 112 lines cut ...]
2022-03-04 08:16:03 DEBUG: Factory Request[ReadHoldingRegistersRequest: 3]
2022-03-04 08:16:03 DEBUG: Frame advanced, resetting header!!
2022-03-04 08:16:03 DEBUG: validate: fc-[3] address-1601: count-23
2022-03-04 08:16:03 DEBUG: getValues fc-[3] address-1601: count-23
2022-03-04 08:16:03 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:03 DEBUG: send: [ReadHoldingRegistersResponse (23)]- b'02032e000000000005000500050005000000000000000f000127102710271000000000000005dc0078000000004e20000065fe'
2022-03-04 08:16:03 DEBUG: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32846, 17743, 32846, 17743, 0, 0, 0, 0, 0, 0, 24946, 17941, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13705, 48852, 0, 0, 0, 0, 0, 0, 42598, 50201, 0, 0, 0, 0, 0, 0, 56372, 17448, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 26152, 17480, 0, 0, 13366, 17745, 0, 0, 0, 0, 0, 0, 0, 0]
2022-03-04 08:16:03 DEBUG: setValues[3] 1101:82
2022-03-04 08:16:04 DEBUG: Getting Frame - 0x3 0x6 0x72 0x0 0x6
2022-03-04 08:16:04 DEBUG: Factory Request[ReadHoldingRegistersRequest: 3]
2022-03-04 08:16:04 DEBUG: Frame advanced, resetting header!!
2022-03-04 08:16:04 DEBUG: validate: fc-[3] address-1651: count-6
2022-03-04 08:16:04 DEBUG: getValues fc-[3] address-1651: count-6
2022-03-04 08:16:04 DEBUG: send: [ReadHoldingRegistersResponse (6)]- b'02030c0000000200040000000000054cd2'
2022-03-04 08:16:05 INFO: return set data set from NMakel
2022-03-04 08:16:05 DEBUG: [27441, 17880, 22036, 17761, 27441, 17880, 22036, 17761, 62259, 50059, 0, 0, 0, 0, 0, 0, 58982, 17260, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 52429, 16967]
2022-03-04 08:16:05 DEBUG: Getting Frame - 0x6 0x6 0x47 0x0 0x0
2022-03-04 08:16:05 DEBUG: Factory Request[WriteSingleRegisterRequest: 6]
2022-03-04 08:16:05 DEBUG: setValues[3] 1001:34
2022-03-04 08:16:05 DEBUG: Frame advanced, resetting header!!
2022-03-04 08:16:05 DEBUG: validate: fc-[6] address-1608: count-1
2022-03-04 08:16:05 DEBUG: setValues[6] 1608:1
2022-03-04 08:16:05 DEBUG: getValues fc-[6] address-1608: count-1
2022-03-04 08:16:05 DEBUG: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 32846, 17743, 32846, 17743, 0, 0, 0, 0, 0, 0, 24946, 17941, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 13705, 48852, 0, 0, 0, 0, 0, 0, 42598, 50201, 0, 0, 0, 0, 0, 0, 56372, 17448, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 26152, 17480, 0, 0, 13366, 17745, 0, 0, 0, 0, 0, 0, 0, 0] 
```

SebuZet, 2022-03-04:
And my current meter congig looks like this:

```
slave_ctx = ModbusSlaveContext()

block_1601 = BinaryPayloadBuilder(byteorder=Endian.Big, wordorder=Endian.Little)
block_1601.add_32bit_int(0) # config passcode
block_1601.add_16bit_int(confparser[meter].getint("ct_current", fallback=default_config["meters"]["ct_current"])) # ct rated current
block_1601.add_16bit_int(confparser[meter].getint("ct_current", fallback=default_config["meters"]["ct_current"])) # ct rated current l1
block_1601.add_16bit_int(confparser[meter].getint("ct_current", fallback=default_config["meters"]["ct_current"])) # ct rated current l2
block_1601.add_16bit_int(confparser[meter].getint("ct_current", fallback=default_config["meters"]["ct_current"])) # ct rated current l3
block_1601.add_16bit_int(confparser[meter].getint("ct_inverted", fallback=default_config["meters"]["ct_inverted"])) # ct direction inversion
block_1601.add_16bit_int(1) # measurement averaging
block_1601.add_16bit_int(0) # power scale
block_1601.add_16bit_int(15) # demand period
block_1601.add_16bit_int(1) # demand subintervals
block_1601.add_16bit_int(10000) # power/energy adjustment l1
block_1601.add_16bit_int(10000) # power/energy adjustment l2
block_1601.add_16bit_int(10000) # power/energy adjustment l3
block_1601.add_16bit_int(0) # ct phase angle adjustment l1
block_1601.add_16bit_int(0) # ct phase angle adjustment l2
block_1601.add_16bit_int(0) # ct phase angle adjustment l3
block_1601.add_16bit_int(1500) # minimum power reading
block_1601.add_16bit_int(confparser[meter].getint("phase_offset", fallback=default_config["meters"]["phase_offset"])) # phase offset
block_1601.add_16bit_int(0) # reset energy
block_1601.add_16bit_int(0) # reset demand
block_1601.add_16bit_int(20000) # current scale
block_1601.add_16bit_int(0) # io pin mode
slave_ctx.setValues(3, 1600, block_1601.to_registers())

block_1651 = BinaryPayloadBuilder(byteorder=Endian.Big, wordorder=Endian.Little)
block_1651.add_16bit_int(0) # apply config
block_1651.add_16bit_int(address) # modbus address
block_1651.add_16bit_int(4) #4 baud rate
block_1651.add_16bit_int(0) # parity mode
block_1651.add_16bit_int(0) # modbus mode
block_1651.add_16bit_int(5) # message delay
slave_ctx.setValues(3, 1650, block_1651.to_registers())

block_1701 = BinaryPayloadBuilder(byteorder=Endian.Big, wordorder=Endian.Little)
block_1701.add_32bit_int(confparser[meter].getint("serial_number", fallback=default_config["meters"]["serial_number"])) # serial number
block_1701.add_32bit_int(0) # uptime (s)
block_1701.add_32bit_int(0) # total uptime (s)
block_1701.add_16bit_int(201) # wattnode model
block_1701.add_16bit_int(31) # firmware version
block_1701.add_16bit_int(0) # wattnode options
block_1701.add_16bit_int(0) # error status
block_1701.add_16bit_int(0) # power fail count
block_1701.add_16bit_int(0) # crc error count
block_1701.add_16bit_int(0) # frame error count
block_1701.add_16bit_int(0) # packet error count
block_1701.add_16bit_int(0) # overrun count
block_1701.add_16bit_int(0) # error status 1
block_1701.add_16bit_int(0) # error status 2
block_1701.add_16bit_int(0) # error status 3
block_1701.add_16bit_int(0) # error status 4
block_1701.add_16bit_int(0) # error status 5
block_1701.add_16bit_int(0) # error status 6
block_1701.add_16bit_int(0) # error status 7
block_1701.add_16bit_int(0) # error status 8
slave_ctx.setValues(3, 1700, block_1701.to_registers())
```

I was playing with some values but without success

SebuZet, 2022-03-04:
I was trying to connect to SE from python3 command prompt.
Is it possible that SE is not handling request if it is set into multi-bus mode?

```
>>> inverter
Inverter(/dev/ttySC0, connectionType.RTU: stopbits=1, parity=N, baud=9600, timeout=1, retries=3, unit=0x1)
>>> inverter.connected()
True
>>>
>>> inverter.read_all()
{}
>>>
```

My SE configuration:
[image]
[image]
[image]