nmakel/solaredge_meterproxy #19: What / where to feed the kWh imported and exported

marcelrv, 2022-03-27:
Hi @nmakel thanks for your solaredge proxy.
I'm feeding info from my Dutch P1 meter to Solaredge. This output provides separate fields for import & export
The current import/export W for solaredge is working fine with calculating export-import.

I'm struggling to understand what the solarEdge expects as values for the export & import KwH
I was feeding the meterposition (`electricityDeliveredT1` + `electricityDeliveredT2` & `electricitySuppliedT1` + `electricitySuppliedT2`)  to it by supplying the meter in `export_energy_active` and `import_energy_active`  but that gives very odd bar in the solaredge app as it seems my total import and export in a single day.

Do you know what SolarEdge expects for this, is it a value that resets every day, or every hour, never (as I was expecting)?

Note this is what I get from my meter (using https://github.com/mcrapts/p1-reader to convert to MQTT)
``` 
{
  "versionInformation": 50,
  "timestamp": 1648374916,
  "equipmentIdentifier": "E0069000717977322",
  "electricityDeliveredT1": 1593648,
  "electricityDeliveredT2": 1337601,
  "electricitySuppliedT1": 148511,
  "electricitySuppliedT2": 297506,
  "tariffIndicator": 1,
  "electricityDeliveredActual": 0,
  "electricitySuppliedActual": 2608,
  "powerFailures": 41,
  "powerFailuresLong": 28,
  "powerFailureEventLog": "10|0-0:96.7.19|220326050040W|0004303445|220326050040W|0004303446|220326050039W|0004303448|220301082909W|0002155955|220301082909W|0002155955|220301082909W|0002155957|220204115736W|0000008462|220204115736W|0000008462|220204115736W|0000008464|220129113651W|0004300285",
  "voltageSagsL1": 11,
  "voltageSagsL2": 10,
  "voltageSagsL3": 6,
  "voltageSwellsL1": 15,
  "voltageSwellsL2": 15,
  "voltageSwellsL3": 16,
  "textMessage": "",
  "instantaneousVoltageL1": 233,
  "instantaneousVoltageL2": 231,
  "instantaneousVoltageL3": 231,
  "instantaneousCurrentL1": 4,
  "instantaneousCurrentL2": 3,
  "instantaneousCurrentL3": 3,
  "instantaneousActivePowerL1Plus": 0,
  "instantaneousActivePowerL2Plus": 0,
  "instantaneousActivePowerL3Plus": 0,
  "instantaneousActivePowerL1Min": 948,
  "instantaneousActivePowerL2Min": 883,
  "instantaneousActivePowerL3Min": 775,
  "deviceType1": "003",
  "equipmentIdentifier1": "G0073004116253431",
  "deviceValue1": 1037.661
}
```

marcelrv, 2022-03-27:
[image]

nmakel, 2022-03-28:
Hi! Here's a sample of the data I'm sending to the inverter (obtained from a SDM630 in single phase mode), which should give you an idea of the expected units:

```
{
    'energy_active': 171.77099609375,                               # kWh (total import + export)
    'import_energy_active': 57.836997985839844,             # kWh
    'power_active': 561.8052368164062,                            # W
    'l1_power_active': 129.37228393554688,
    'l2_power_active': 344.997314453125,
    'l3_power_active': 88.22502136230469,
    'voltage_ln': 238.35206604003906,                               # V
    'l1n_voltage': 238.43624877929688,
    'l2n_voltage': 238.165771484375,
    'l3n_voltage': 238.48741149902344,
    'voltage_ll': 0.0,
    'l12_voltage': 0.0,
    'l23_voltage': 0.0,
    'l31_voltage': 0.0,
    'frequency': 49.8844108581543,                                    # Hz
    'l1_energy_active': 171.77099609375,                            # kWh (import + export)
    'l1_import_energy_active': 57.836997985839844,          # kWh
    'export_energy_active': 113.93399810791016,               # kWh
    'l1_export_energy_active': 113.93399810791016,
    'energy_reactive': 153.6859893798828,
    'l1_energy_reactive': 153.6859893798828,
    'energy_apparent': 230.48785400390625,
    'l1_energy_apparent': 230.48785400390625,
    'power_factor': 0.6266472935676575,
    'l1_power_factor': 0.5347498059272766,
    'l2_power_factor': 0.6997311115264893,
    'l3_power_factor': 0.5260195136070251,
    'power_reactive': -698.6649169921875,
    'l1_power_reactive': -204.4337921142578,
    'l2_power_reactive': -352.2327880859375,
    'l3_power_reactive': -142.6428985595703,
    'power_apparent': 896.5253295898438,
    'l1_power_apparent': 241.93048095703125,
    'l2_power_apparent': 493.0426330566406,
    'l3_power_apparent': 167.72193908691406,
    'l1_current': 1.0146546363830566,
    'l2_current': 2.0701656341552734,
    'l3_current': 0.7032737731933594,
    'demand_power_active': 390.0171203613281,
    'maximum_demand_power_active': 0.0,
    'demand_power_apparent': 802.758544921875,
    'l1_demand_power_active': 247.37389027341123,
    'l2_demand_power_active': 399.4080980680883,
    'l3_demand_power_active': 166.98351734959851
}
```

nmakel, 2022-03-28:
> Here you see in my week consumption 4.05 MWh which is a biiiiit too much ...
> 
> [image]

So this is probably a unit issue. I also ran into this accidentally during debugging. To remove this value from the dashboard I needed to open a ticket with SolarEdge support. With a bit of luck they can remove the erroneous values from the graphs.

marcelrv, 2022-04-07:
Thanks for the input.
I'm running the script now for couple of days. It happened 2 times more that it had and will indeed create a support ticket for removing those.
Still awaiting the day I see matching numbers for energy in the portal . (power is working fine). 
The strange thing is that I see the correct values for the energy on the powerEdge itself. (at least I see the total meter kWh for import & export)

What I noticed is that for me the timeout settings have a very big impact. I somewhat expect that the strange numbers are cuz of data errors when I was setting various timeouts
with timeout 1sec, the device get stuck on requesting address-1601
with timeout .1 I see requests for 1601, 1651,1701, 1001, 1101,2128 and many 1011
every seconds there are quite a few request for 1011, than every 4 sec the others are requested.

Is it normal to have so many 1011 requests (I see around 8 requests/second) or is it some sort of error and it is not happening with a regular wattnode?

What is unexpected / strange in your example is the energy_active ,  as acc to the spec I was assuming it is the `export - import`, but in your example you have it as `import+export `

marcelrv, 2022-04-07:
Note, I assume the Energy shown in the screen is suppose to show the current meter value, not the daily number right?

I'm feeding
```
 {
   "demand_power_active": 801.6722222222222,
    "energy_active": 2536.827,
    "export_energy_active": 547.341,
    "frequency": 50,
    "import_energy_active": 3084.168,
    "l1_demand_power_active": 44.044444444444444,
    "l1_export_energy_active": 547.341,
    "l1_import_energy_active": 3084.168,
    "l1_power_active": 37.0,
    "l1n_voltage": 234.0,
    "l2_demand_power_active": 345.21666666666664,
    "l2_power_active": 351.0,
    "l2n_voltage": 233.0,
    "l3_demand_power_active": 411.8333333333333,
    "l3_power_active": 446.0,
    "l3n_voltage": 233.0,
    "p1_current": 0.1581196581196581,
    "p2_current": 1.5064377682403434,
    "p3_current": 1.9141630901287554,
    "power_active": 839.0,
    "voltage_ln": 234.0
}
```
[image]

marcelrv, 2022-04-09:
I can say that I finally had 2 days that everything fully matched, import & export included. 
The trick was the setting of  `timeout = .1` .

dunno if the rootcause of the issue, if it is something in the hardware, or the pymodbus or something in the solarEdge that causes the behavior.

[image]

[image]

nmakel, 2022-04-11:
> Note, I assume the Energy shown in the screen is suppose to show the current meter value, not the daily number right?
> 
> I'm feeding...

Check your `p1_current` `p2_current` and `p3_current`. Those should be `l1_current` etc.

My unit does not have a display, so I cannot confirm, but it sounds right.

nmakel, 2022-04-11:
> I can say that I finally had 2 days that everything fully matched, import & export included. The trick was the setting of `timeout = .1` .
>
> dunno if the rootcause of the issue, if it is something in [...]

Great news, thanks for letting me know! The timeout value is good to know, I will add that as the default -- my setup is based on RS485 to ethernet converters, so I never have this issue.

nmakel, 2022-04-14:
Are you using the latest solaredge firmware, by the way? Just to confirm this still works on newer firmware -- I have mine locked to an older release.

marcelrv, 2022-04-15:
Before installing the energy meter I updated to the latest firmware for inverters with display: Version Date July 2020
Single Phase Inverter with HD-Wave Technology | CPU  3.2537 |  DSP1 1.0.1036 | DSP2 2.0.910

So, yes it still works.

onsmam, 2022-05-07:
nice work, i also want to try to connect my P1 meter via mqtt, but i cannot seem to find what values has to be mapped to which channel could you provide your config? where which channel is which telegram topic, 

i now use dsmr reader, which wil provide some json topics, not all in 1 post divided in several post. is this a problem? so in 1 post the day consumption will be posted, and in another post the actual power values will be posted.

and is it needed that the total import/export need to be supplied? or is day import and a day export value ok?

thanx!

marcelrv, 2024-01-01:
closing as it works perfect now

closed 2024-01-01 by marcelrv