nmakel/solaredge_meterproxy #28: semp-tcp.py  issue

git3smurf, 2023-05-13:
When I run semp-tcp.py i get this error: 

root@SEmeterproxy:/solaredge_meterproxy# python3 semp-tcp.py -v
2023-05-13 10:56:28 WARNING: No meters defined in semp-tcp.conf
Traceback (most recent call last):
  File "semp-tcp.py", line 256, in <module>
    server = StartTcpServer(
TypeError: StartTcpServer() got an unexpected keyword argument 'framer'
root@SEmeterproxy:/solaredge_meterproxy# 

this is my configuration file (semp-tcp.conf) 
server]
# Serving IP address.
# optional, default: all interfaces
address = "192.168.40.70"

# Serving port.
# optional, default: 5502
#port = 4196

# Modbus frame type, set to rtu for Modbus RTU over TCP
# optional, default: socket
#framer = socket
framer = "rtu"

# Logging level, CRITICAL, ERROR, WARNING, INFO, DEBUG
# optional, default: INFO
log_level = DEBUG

# Masqueraded meters, comma separated.
# optional, default: ''
#meters = meter1, meter2 

# Meters defined in [server] need a config section, one per meter.
# Depending on the type of meter that is to be masqueraded, you can
# define a number of generic and type specific variables.

# Modbus address of the meter as defined in the SolarEdge inverter.
# This value needs to be unique.
# optional, default: 2
#dst_address = 2

# Source meter type, which corresponds to a script in /devices.
# The generic.py device returns null values.
# optional, default: generic
#type = generic

# Masqueraded serial number.
# Need not be correct, must be unique, must be an integer.
# optional, default: 987654
#serial_number = 987654

# Current transformer amperage rating.
# optional, default: 5
ct_current = 10

# Current transformer direction inversion, set to 1 if required.
# optional, default: 0
#ct_inverted = 0

# Offset between phases, set to 0, 90, 120 or 180.
# optional, default: 0
#phase_offset = 120

# Number of seconds between value refreshes.

It also mentions "no meters defined" while on default ...  but when I set meters = meter1 in semp-tcp.conf  I get this...  

root@SEmeterproxy:/solaredge_meterproxy# python3 semp-tcp.py -v
Traceback (most recent call last):
  File "semp-tcp.py", line 153, in <module>
    address = confparser[meter].getint("dst_address", fallback=default_config["meters"]["dst_address"])
  File "/usr/lib/python3.8/configparser.py", line 960, in __getitem__
    raise KeyError(key)
KeyError: 'meter1'
root@SEmeterproxy:/solaredge_meterproxy# 

Then it hangs on line 153 ...

git3smurf, 2023-05-23:
I managed to correct the meter problem...  by just completing the meterdata in the config file... But i still have errors which I cannot resolve....  

This is what the log shows now:

`root@SEmeterproxy:/solaredge_meterproxy# python3 semp-tcp.py -vvv
2023-05-23 09:30:12 DEBUG: Started MQTT connection to server - topic: 192.168.40.85:1883  - meter
2023-05-23 09:30:12 DEBUG: [1234, 0, 10, 10, 10, 10, 0, 0, 0, 15, 1, 10000, 10000, 10000, 64536, 64536, 64536, 1500, 120, 0, 0, 20000, 0]
2023-05-23 09:30:12 DEBUG: setValues[3] address-1601: count-23
2023-05-23 09:30:12 DEBUG: [0, 2, 4, 0, 0, 5]
2023-05-23 09:30:12 DEBUG: setValues[3] address-1651: count-6
2023-05-23 09:30:12 DEBUG: [4614, 15, 0, 0, 0, 0, 202, 31, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2023-05-23 09:30:12 DEBUG: setValues[3] address-1701: count-23
2023-05-23 09:30:12 INFO: Created <Thread(t_update_2, initial)>: meter1 mqttP1 {'client': <paho.mqtt.client.Client object at 0x7f1ddc307e80>, 'host': '192.168.40.85', 'port': 1883, 'keepalive': 60, 'meterValuesTopic': 'meter', 'willTopic': 'will', 'willMsg': 'MeterProxy Disconnected'}
2023-05-23 09:30:12 INFO: MQTT connected to 192.168.40.85:1883 - topic: 'meter' with result code 0.
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 3591, in _thread_main
    self.loop_forever(retry_first_connection=True)
  File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 1756, in loop_forever
    rc = self._loop(timeout)
  File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 1164, in _loop
    rc = self.loop_read()
  File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 1556, in loop_read
    rc = self._packet_read()
  File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 2439, in _packet_read
    rc = self._packet_handle()
  File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 3033, in _packet_handle
    return self._handle_publish()
  File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 3327, in _handle_publish
    self._handle_on_message(message)
  File "/usr/local/lib/python3.8/dist-packages/paho/mqtt/client.py", line 3570, in _handle_on_message
    on_message(self, self._userdata, message)
  File "/solaredge_meterproxy/devices/mqttP1.py", line 70, in on_message
    lastValues ['demand_power_active'] = demandAvg.next ( lastValues['powerImportedActual'] - lastValues['powerExportedActual'] )
KeyError: 'powerImportedActual'
2023-05-23 09:30:13 CRITICAL: t_update_2: 'instantaneousVoltageL1'
2023-05-23 09:30:13 INFO: Starting <Thread(t_update_2, started 139766218807040)>
Traceback (most recent call last):
  File "semp-tcp.py", line 256, in <module>
    server = StartTcpServer(
TypeError: StartTcpServer() got an unexpected keyword argument 'framer'
root@SEmeterproxy:/solaredge_meterproxy# `

This is my configuration file: 

`GNU nano 4.8                        semp-tcp.conf                                  
[server]
# Serving IP address.
# optional, default: all interfaces
address = 192.168.40.81

# Serving port.
# optional, default: 5502
port = 4196

# Modbus frame type, set to rtu for Modbus RTU over TCP
# optional, default: socket
#framer = socket
framer = rtu

# Logging level, CRITICAL, ERROR, WARNING, INFO, DEBUG
# optional, default: INFO
log_level = DEBUG

# Masqueraded meters, comma separated.
# optional, default: ''
meters = meter1

[meter1]
type = mqttP1
host = 192.168.40.85
#port = 1883
src_address=1
dst_address=2
meterValuesTopic = meter
willTopic = will
refresh_rate = 15
#serial_number = "0x0D010552"

# Meters defined in [server] need a config section, one per meter.
# Depending on the type of meter that is to be masqueraded, you can
# define a number of generic and type specific variables.

# Modbus address of the meter as defined in the SolarEdge inverter.
# This value needs to be unique.
# optional, default: 2
#dst_address = 2

# Source meter type, which corresponds to a script in /devices.
# The generic.py device returns null values.
# optional, default: generic
#type = generic

# Masqueraded serial number.
# Need not be correct, must be unique, must be an integer.
# optional, default: 987654
#serial_number = 987654

# Current transformer amperage rating.
# optional, default: 5
ct_current = 10

# Current transformer direction inversion, set to 1 if required.
# optional, default: 0
#ct_inverted = 0

# Offset between phases, set to 0, 90, 120 or 180.
# optional, default: 0
#phase_offset = 120

# Number of seconds between value refreshes.
# optional, default: 5
#refresh_rate = 5`

Between the SolarEdge SE6K and the solaredge_meterproxy I have a Waveshare Gateway with this configuration:
[image]

cratoo, 2024-05-30:
@git3smurf did you get this running and if yes how did you solve your issues and how did you configure the SE6K in regards to rs485?