nmakel/solaredge_meterproxy #29: Troublesome Own Usage (Eigen Verbruik)and Export/Bought (Exporteren/Gekocht) reporting

fkemps, 2024-01-18:
Hello SE geeks,

I'm having trouble reporting my Own Usage (Eigen Verbruik) and Export/Bought (Exporteren/Gekocht) on the SE Monitoring platform.  Using the Solaredge_meter with MQTT connected to SE5000H via modbus RS485-1 (termination on).

The solar **panel**, **home**, and **export** power seem to work well, both online monitoring platform and SolarEdge App are showing the correct kW values and green/orange arrows during the day.
I can't get the reporting working correctly (blue/green and blue/red bars) because the Exporteren and Gekocht values are incorrect for unknow reason.

[image]

Questions
1) Is the modbus communication working as expected?
2) DEBUG information looks different compared to what I've found online. Why the "**Frame advanced, resetting header!!**" message.
3) What are the necessary MQTT fields and value dimensions?
4) Am I correct that latest pymodbus version supported is 2.5.3? (with latest 3.6.3 solardedge_meterproxy didn't start)

Communication between solaredge_meterproxy and SE5000H is via Elfin EW11A, 9600_8,_N,_1 as shown.

```script
Jan 19 00:15:38 hab systemd[1]: Started SolarEdge Meter daemon simulating meter device.
Jan 19 00:15:39 hab python3[21371]: 2024-01-19 00:15:39 DEBUG: [1234, 0, 5, 5, 5, 5, 0, 0, 0, 15, 1, 10000, 10000, 10000, 64536, 64536, 64536, 1500, 120, 0, 0, 20000, 0]
Jan 19 00:15:39 hab python3[21371]: 2024-01-19 00:15:39 DEBUG: setValues[3] 1601:23
Jan 19 00:15:39 hab python3[21371]: 2024-01-19 00:15:39 DEBUG: [0, 2, 4, 0, 0, 5]
Jan 19 00:15:39 hab python3[21371]: 2024-01-19 00:15:39 DEBUG: setValues[3] 1651:6
Jan 19 00:15:39 hab python3[21371]: 2024-01-19 00:15:39 DEBUG: [4614, 15, 0, 0, 0, 0, 202, 31, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
Jan 19 00:15:39 hab python3[21371]: 2024-01-19 00:15:39 DEBUG: setValues[3] 1701:23
Jan 19 00:15:39 hab python3[21371]: 2024-01-19 00:15:39 INFO: Created <Thread(t_update_2, initial)>: meter1 mqtt {'client': <paho.mqtt.client.Client object at 0x760f3058>, 'host': '192.168.254.13', 'port': 1883, 'keepalive': 60, 'meterValuesTopic': 'energy/meter', 'willTopic': 'energy/will', 'willMsg': 'MeterProxy Disconnected'}
Jan 19 00:15:39 hab python3[21371]: 2024-01-19 00:15:39 INFO: Connected to MQTT: 192.168.254.13:1883/energy/meter
Jan 19 00:15:40 hab python3[21371]: 2024-01-19 00:15:40 DEBUG: MQTT message received: {
Jan 19 00:15:40 hab python3[21371]: "import_energy_active": 0.372,
Jan 19 00:15:40 hab python3[21371]: "l1_power_active": 16.000,
Jan 19 00:15:40 hab python3[21371]: "l2_power_active": 198.000,
Jan 19 00:15:40 hab python3[21371]: "l3_power_active": 158.000,
Jan 19 00:15:40 hab python3[21371]: "voltage_ln": 235.7,
Jan 19 00:15:40 hab python3[21371]: "l1n_voltage": 235.7,
Jan 19 00:15:40 hab python3[21371]: "l2n_voltage": 235,
Jan 19 00:15:40 hab python3[21371]: "l3n_voltage": 233.5,
Jan 19 00:15:40 hab python3[21371]: "frequency": 50,
Jan 19 00:15:40 hab python3[21371]: "l1_energy_active": 0.016,
Jan 19 00:15:40 hab python3[21371]: "l2_energy_active": 0.198,
Jan 19 00:15:40 hab python3[21371]: "l3_energy_active": 0.158,
Jan 19 00:15:40 hab python3[21371]: "l1_import_energy_active": 0.016,
Jan 19 00:15:40 hab python3[21371]: "l2_import_energy_active": 0.198,
Jan 19 00:15:40 hab python3[21371]: "l3_import_energy_active": 0.158,
Jan 19 00:15:40 hab python3[21371]: "export_energy_active": 0,
Jan 19 00:15:40 hab python3[21371]: "l1_export_energy_active": 0,
Jan 19 00:15:40 hab python3[21371]: "l2_export_energy_active": 0,
Jan 19 00:15:40 hab python3[21371]: "l3_export_energy_active": 0,
Jan 19 00:15:40 hab python3[21371]: "l1_current": 0.06788290,
Jan 19 00:15:40 hab python3[21371]: "l2_current": 0.84255319,
Jan 19 00:15:40 hab python3[21371]: "l3_current": 0.67665953
Jan 19 00:15:40 hab python3[21371]: }
Jan 19 00:15:40 hab python3[21371]: 2024-01-19 00:15:40 DEBUG: [0, 0, 30409, 16062, 0, 0, 30409, 16062, 0, 0, 0, 16768, 0, 17222, 0, 17182, 45875, 17259, 45875, 17259, 0, 17259, 32768, 17257, 0, 0, 0, 0, 0, 0, 0, 0, 0, 16968]
Jan 19 00:15:40 hab python3[21371]: 2024-01-19 00:15:40 INFO: Starting <Thread(t_update_2, started 1965028416)>
Jan 19 00:15:40 hab python3[21371]: 2024-01-19 00:15:40 DEBUG: setValues[3] 1001:34
Jan 19 00:15:40 hab python3[21371]: 2024-01-19 00:15:40 DEBUG: [4719, 15491, 49283, 15946, 51905, 15905, 4719, 15491, 49283, 15946, 51905, 15905, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1585, 15755, 45457, 16215, 14735, 16173, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
Jan 19 00:15:40 hab python3[21371]: 2024-01-19 00:15:40 DEBUG: setValues[3] 1101:82
Jan 19 00:15:42 hab python3[21371]: 2024-01-19 00:15:42 DEBUG: Started thread to serve client at ('192.168.254.109', 9999)
Jan 19 00:15:42 hab python3[21371]: 2024-01-19 00:15:42 DEBUG: Client Connected [192.168.254.109:9999]
Jan 19 00:15:42 hab python3[21371]: 2024-01-19 00:15:42 DEBUG: Handling data: 0x2 0x3 0x0 0x1 0x0 0x1 0xd5 0xf9
Jan 19 00:15:42 hab python3[21371]: 2024-01-19 00:15:42 DEBUG: Getting Frame - 0x3 0x0 0x1 0x0 0x1
Jan 19 00:15:42 hab python3[21371]: 2024-01-19 00:15:42 DEBUG: Factory Request[ReadHoldingRegistersRequest: 3]
Jan 19 00:15:42 hab python3[21371]: 2024-01-19 00:15:42 DEBUG: Frame advanced, resetting header!!
Jan 19 00:15:42 hab python3[21371]: 2024-01-19 00:15:42 DEBUG: validate: fc-[3] address-2: count-1
Jan 19 00:15:42 hab python3[21371]: 2024-01-19 00:15:42 DEBUG: getValues fc-[3] address-2: count-1
Jan 19 00:15:42 hab python3[21371]: 2024-01-19 00:15:42 DEBUG: send: [ReadHoldingRegistersResponse (1)]- b'0203020000fc44'
Jan 19 00:15:42 hab python3[21371]: 2024-01-19 00:15:42 DEBUG: Handling data: 0x2 0x3 0x3 0xf2 0x0 0x6 0x64 0x4c
Jan 19 00:15:42 hab python3[21371]: 2024-01-19 00:15:42 DEBUG: Getting Frame - 0x3 0x3 0xf2 0x0 0x6
Jan 19 00:15:42 hab python3[21371]: 2024-01-19 00:15:42 DEBUG: Factory Request[ReadHoldingRegistersRequest: 3]
Jan 19 00:15:42 hab python3[21371]: 2024-01-19 00:15:42 DEBUG: Frame advanced, resetting header!!
Jan 19 00:15:42 hab python3[21371]: 2024-01-19 00:15:42 DEBUG: validate: fc-[3] address-1011: count-6
[... 474 lines cut ...]
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: Frame advanced, resetting header!!
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: validate: fc-[3] address-1011: count-6
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: getValues fc-[3] address-1011: count-6
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: send: [ReadHoldingRegistersResponse (6)]- b'02030c00004180000043470000431ae8a1'
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: Handling data: 0x2 0x3 0x4 0x4c 0x0 0x52 0x4 0xe3
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: Getting Frame - 0x3 0x4 0x4c 0x0 0x52
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: Factory Request[ReadHoldingRegistersRequest: 3]
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: Frame advanced, resetting header!!
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: validate: fc-[3] address-1101: count-82
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: getValues fc-[3] address-1101: count-82
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: send: [ReadHoldingRegistersResponse (82)]- b'0203a4126f3c83c6a83e4bb22d3e1d126f3c83c6a83e4bb22d3e1d00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000f7183d8a99413f58b1e53f2800000000000000000000000000000000000000000000000000000000f991'
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: Handling data: 0x2 0x3 0x3 0xf2 0x0 0x6 0x64 0x4c
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: Getting Frame - 0x3 0x3 0xf2 0x0 0x6
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: Factory Request[ReadHoldingRegistersRequest: 3]
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: Frame advanced, resetting header!!
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: validate: fc-[3] address-1011: count-6
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: getValues fc-[3] address-1011: count-6
Jan 19 00:15:52 hab python3[21371]: 2024-01-19 00:15:52 DEBUG: send: [ReadHoldingRegistersResponse (6)]- b'02030c00004180000043470000431ae8a1'
```

My environment and configuration details can be found in the drawing

[image]

Thanks in advance for any guidance to get this last step working........in preparation for adding a SE battery to the environment.

nmakel, 2024-02-12:
The debug output looks fine. As you see the meter is correctly registered and reporting values in the SolarEdge app. 

__What are the necessary MQTT fields and value dimensions?__ You'll need to ask @marcelrv as you're using his P1 scripts.

__Am I correct that latest pymodbus version supported is 2.5.3? (with latest 3.6.3 solardedge_meterproxy didn't start)__ Grab the latest github version, which should support newer pymodbus versions.

The screenshot you posted shows correct self-consumption, or am I misinterpreting it?

marcelrv, 2024-02-12:
Sorry I'm on holidays this week.. 
Can take a look next week...
But the screenshot indeed looks like it has the right consumption and delivery to the net

fkemps, 2024-02-12:
Hello nmakel,

Upgraded to pymodbus 3.6.3 but solaredge_meterproxy did not start anymore with following error.

```
Feb 12 21:39:55 hab python3[21214]: Traceback (most recent call last):
Feb 12 21:39:55 hab python3[21214]:   File "/etc/openhab/github/solaredge_meterproxy/semp-tcp.py", line 160, in <module>
Feb 12 21:39:55 hab python3[21214]: 2024-02-12 21:39:55 INFO: Connected to MQTT: 192.168.254.13:1883/energy/meter
Feb 12 21:39:55 hab python3[21214]:     block_1601 = BinaryPayloadBuilder(byteorder=Endian.Big, wordorder=Endian.Little)
Feb 12 21:39:55 hab python3[21214]:   File "/usr/lib/python3.9/enum.py", line 405, in __getattr__
Feb 12 21:39:55 hab python3[21214]:     raise AttributeError(name) from None
Feb 12 21:39:55 hab python3[21214]: AttributeError: Big
```
Solved it by updating semp-tcp.py and replacing **Big** to **BIG** and **Little** to **LITTLE** e.g.`block_1601 = BinaryPayloadBuilder(byteorder=Endian.Big, wordorder=Endian.Little)` by block_1601 = BinaryPayloadBuilder(byteorder=Endian.BIG, wordorder=Endian.LITTLE).

I suggest updating your code as I've seen a couple of posts where ppl reported solaredge_meterproxy failed to start with the same error.

I'm not using marcelrv's mqttP1.py device but the default mqtt device and feed it from OpenHAB via a rule in which I publish values to MQTT.
I've been busy for a month now but can't get it stable on Solaredge Monitoring and values incorrect.

- There are gaps during the day in the monitoring, the reported EigenVerbruik is way to high.
- The live PV-production, Consumption and Export values are correct.
- Looks like the SE modbus is having intermittent issues as OpenHab loses connectivity 1 or 2 times a day for a few seconds (unreachable).

The SE monitoring behavior changes with the values I publish on MQTT.
**_That's why I'm very curious about what needs to be published on MQTT and in what values._**

This is what I have currently

**Consuming from net only.**
```
{
"l1_power_active": 16.000,
"l2_power_active": 220.00,
"l3_power_active": 242.000,
"voltage_ln": 236.8,
"l1n_voltage": 236.8,
"l2n_voltage": 236.5,
"l3n_voltage": 235.1,
"frequency": 50,
"l1_energy_active": 0.016,
"l2_energy_active": 0.22,
"l3_energy_active": 0.242,
"l1_import_energy_active": 0.016,
"l2_import_energy_active": 0.22,
"l3_import_energy_active": 0.242,
"l1_export_energy_active": 0,
"l2_export_energy_active": 0,
"l3_export_energy_active": 0,
"export_energy_active": 0,
"l1_current": 0.06756757,
"l2_current": 0.93023256,
"l3_current": 1.02934921
}
```

**Consuming and delivering to net.**
```
{
"energy_active": 1.083,
"l1_power_active": -194.000,
"l2_power_active": 265.000,
"l3_power_active": 1012.000,
"voltage_ln": 237.1,
"l1n_voltage": 237.1,
"l2n_voltage": 236.5,
"l3n_voltage": 234.3,
"frequency": 50,
"l1_energy_active": -0.194,
"l2_energy_active": 0.265,
"l3_energy_active": 1.012,
"l1_import_energy_active": 0,
"l2_import_energy_active": 0.265,
"l3_import_energy_active": 1.012,
"l1_export_energy_active": 0.194,
"l2_export_energy_active": 0,
"l3_export_energy_active": 0,
"export_energy_active": 0,
"l1_current": -0.81822016,
"l2_current": 1.12050740,
"l3_current": 4.31924883
}
``` 

[image]

[image]

[image]

fkemps, 2024-02-14:
Thanks for sharing the SDM630 value details. It gives me something to chew on.

The upgrade of pymodbus v2.5.x to v3.6.x made a huge difference in monitoring stability on the SE monitoring portal.
There are no more gaps during the day, which happened randomly and frequently. This resulted in the high-energy reported peaks (40kWh and higher) as shown in the screenshot. 

This looks promising going forward. 
Today not much PV production and values look much better. Will optimize MQTT shared fields in the next few days and share the result.

[image]

fkemps, 2024-02-21:
With regards to the *_energy_* would it mean that the following calculation should be performed while reporting every 10 seconds?

(W x hrs)/1000 so 500W becomes (500W x 0,00277778h)/1000=0,0138889kWh ?

marcelrv, 2024-02-21:
no, you need report your meter value. ('meterstand') in energy
If you have dual tarif, you will need to sum the value of the day and night meters.

Currently I have similar calulation to find the phase meter values, as the Dutch P1 meter does not provide the meter value per phase, but gives the power  and I wanted to know on which phase I'm consuming the most (so my home battery can be connected there).
In that case I do something like:  Ph1 kWh =  last Ph1 value kWh + ph1 W * (second since last value) / 3600 for each phase.

NB. As SolarEdge is not necessary handling the initial value of the meters very well (see one of my issues earlier in this repo) I changed my script to start at 0 from when I started measuring.  So I always substract the a fixed inital meter value from the meterstand. Otherwise you see an enormous first / initial value. That is disturbing your year totals.

fkemps, 2024-02-21:
Thanks Marcel,
Understood to change the **energy** values to sum of Tarif1 + Tarif2, those are available.

Trying to understand when you're starting with value 0, know why you do it though. Seen these high energy spikes disturbing my year reporting :-(
I'm pushing the values to MQTT from an OpenHab rules, so I guess it would be on (re)starting OpenHab.

This is btw what my Landis Gyr E360 meter spits out on P1.
```
1-3:0.2.8(50)
0-0:1.0.0(231xxx3840W)
0-0:96.1.1(4530303532303xxx35383039343139)
1-0:1.8.1(007777.144*kWh)  <== this would be needed for energy reporting
1-0:1.8.2(005576.970*kWh)  <== this would be needed for energy reporting
1-0:2.8.1(004030.973*kWh)  <== this would be needed for energy reporting
1-0:2.8.2(009081.937*kWh)  <== this would be needed for energy reporting
0-0:96.14.0(0002)
1-0:1.7.0(00.393*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00008)
0-0:96.7.9(00003)
1-0:99.97.0(2)(0-0:96.7.19)(000101000000W)(0000000399*s)(210609060051S)(0000000443*s)
1-0:32.32.0(00024)
1-0:52.32.0(00018)
1-0:72.32.0(00020)
1-0:32.36.0(00001)
1-0:52.36.0(00000)
1-0:72.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(230.5*V)
1-0:52.7.0(231.3*V)
1-0:72.7.0(232.7*V)
1-0:31.7.0(001*A)
1-0:51.7.0(001*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.195*kW)
1-0:41.7.0(00.232*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.034*kW)
!8869
/XMX5LGF0010xxxx4858094
```

marcelrv, 2024-02-21:
Yes, indeed those are the ones I use.

Yes... I also have my first year with a huge spike of 30 MWh ;-) in a single day ( and rest of the year unreadable as the table automatically adjusts to the high value...). I called with SolarEdge support if they could remove it, but the said the couldn't

So this year I when I needed to replace my Solaredge inverter, I made indeed the offset to not have the same problem again.

fkemps, 2024-02-22:
Big thank you nmakel and marcelrv,
After the suggested engery-related updates the "Verbruik" and "Productie" widgets seem to work well.

[image]

Added the per phase energy consumption to get consumption insights as well.

May I ask you about your experience with the battery (Solar Edge 10kW?)
Okay to do this per DM (and in Dutch).

marcelrv, 2024-02-22:
I have no experience with battery... Yet.
I have this week received my battery, ( not SE), but have to see if there is any chance to die it in the SolarEdge monitoring.

But you can reach out, no problem. (You can dm in  in openhab forum, or send email)

Maikel-K, 2024-02-22:
Uploading battery data to Solaredge would be nice!

Got my 90 kWh pack up and running with Victron and would like to tie it into Solaredge too.

cratoo, 2024-06-22:
@marcelrv @Maikel-K did you ever manage to integrate a DIY battery into the SE monitoring?

Maikel-K, 2024-06-22:
No I haven't invested any time in it yet😅

marcelrv, 2024-06-22:
Same here... Still on my wish list... But no time invested

cratoo, 2024-06-22:
Thanks for your response guys. I actually assume that it is not possible anyway, as nothing besides Energy and Power is available for the dummy meter at the SE portal if you check the charts page. Have the impression that the need to configure it as a 'meter' limits the capabilities the inverter would check, even if correct data would be provided. A pity, but great work by @nmakel for this piece of software nonetheless, huge thank you!

cratoo, 2024-08-31:
@fkemps my power reporting now runs since some months without any issues, but I still did not get the energy stuff into SE. Currently my json, that is submitted with semp-tcp.py looks like this:
```
{ "l3_power_active":361.5,
   "l3_energy_active":2.1999999999999993,
    "energy_active":2.1999999999999993
}
```
Am I missing anything? Do I need to add the remaining phases even if it is a single phase system and if this single phase information worked with power?