nmakel/solaredge_meterproxy #35: inside: Setup Guide for external production and mqtt source

macruch2, 2025-03-22:
# Setup Guide for external production and mqtt source

This guide explains how to setup the project in order to simulate an SolarEdge Meter providing data for external production. The advantage is

- you save money
- you do not need to have enough space for installation on DIN rail
- regardless of the physical wiring of you inverter(s)
- more than on inverters on different circuits and phases can "inject" there production data in one software meter

The meter might also be used for other purpose as well, like consumption, production, import/export and so on. This guide assumes an exiting physical meter for import/export and another non SolarEdge inverter from which you would like to provide the production power so that the monitoring portal show e.g. the correct value for self consumption.

### Requirements

- USB to RS485 adapter, like Waveshare USB to RS485 (B) (CH343G-Version) 
    - Waveshare has 3 contacts and seams to be terminated
- Device with USB and capable of running Python, like Raspberry PI or Oranage PI 
    - Operation system already running. This guide assumes linux
    - Python installed
    - Git installed
    - MQTT broker installed
- Twisted Pair cable with at least 3 wires
- data of your other production inverters

### Schematic

[image]

### Steps

1. Download the project code  
    ```
    git clone https://github.com/nmakel/solaredge_meterproxy.git
    cd solaredge_meterproxy
    ```
2. create a python virtual environment and active it  
    ```
    python3 -m venv venv
    source venv/bin/activate
    ```
3. adjust the requirements.txt as some newer versions are not working anymore with the project  
    ```
    pymodbus==3.4.1
    influxdb>=5.3.0
    requests>=2.23.0
    sdm_modbus==0.6.0
    pyserial-asyncio>=0.6.0
    ```
4. install the requirements  
    ```
    pip install -r requirements.txt
    ```
5. plug in your USB adapter and figure out the device, like `/dev/ttyUSB0` or `/dev/ttyACM0`
6. change the config file semp-rtu.conf using your adapter and mqtt  
    ```
    [server]
    # Serving serial device, connected to the SolarEdge inverter.
    # optional, default: /dev/ttyUSB0
    device = /dev/ttyACM0
    
    # Serving serial baud rate.
    # optional, default: 9600
    #baud = 9600
    
    # Parity setting, N, E or O
    # optional, default: E
    #parity = E
    parity = N
    
    # Serving serial timeout, depends on line speed.
    # optional, float, default: 1
    #timeout = 1
    
    # Logging level, CRITICAL, ERROR, WARNING, INFO, DEBUG
    # optional, default: INFO
    #log_level = INFO
    
    # Masqueraded meters, comma separated.
    # optional, default: ''
    #meters = meter1, meter2
    meters = meter1
    
    # Meters defined in [server] need a config section, one per meter.
    # Depending on the type of meter that is to be masqueraded, you can
    # define a number of generic and type specific variables.
    [meter1]
    # Modbus address of the meter as defined in the SolarEdge inverter.
    # This value needs to be unique.
    # optional, default: 2
    dst_address = 3
    
    # Source meter type, which corresponds to a script in /devices.
    # The generic.py device returns null values.
    # optional, default: generic
    type = mqtt
    host = 192.168.xx.xx
    port = 1883
    willTopic = /lastwill
    
    # Masqueraded serial number.
    # Need not be correct, must be unique, must be an integer.
    # optional, default: 987654
    #serial_number = 987654
    
    # Current transformer amperage rating.
    # optional, default: 5
    #ct_current = 50
    ```
7. extend the wiring from your inline meter as shown in the scematic. The Waveshare adapter seems to be terminated so put it at the end of the chain. It's better to shut down your inverter during this work.
8. bring everything back online
9. start the script `python3 semp-rtu.py`
10. connect to your inverters wifi. If you have the solar edge app you can to the menu and choose the inverter mode.   
    [image]
    After this you can start your browser and go to 172.16.0.1.

11. go to Site Communication  
    [image]

12. go to RS485-1  
    [image]

13. Add Modbus Device (Energy Meter)     
    [image]

14. you might have to change the Device ID according to you configuration and change the Meter Function to External Production. You have also to change the Meter Protocoll to WattNode  
    [image]

15. In you current Meter, change the Modbus Termination to OFF  
    [image]

16. After this when you view the status you should see  
    [image]
when you send Data over mqtt you should see the Total Power

17. send your engery data over mqtt on the configured topic. The minimal payload could look like this  
    ```json
    {
      "power_active":0,
      "engery_active":4864.978,
      "l1_energy_active":4864.978,
      "l1_power_active":0
    }
    ```
    
    power is the current power of all your inverters in Watt. Energy is the total produced energy over time in kWh
18. when you view the meter status in the app or on your inverter you should see the injected power value. From this time on the external production is shown together with the solar edge production
19. all other json value you can fill out can be found here https://github.com/nmakel/solaredge\_meterproxy/blob/master/devices/mqtt.py
20. how can now create a service for running and auto starting the meter proxy

menloperk, 2025-03-22:
Thanks, how did you implement step 17?

macruch2, 2025-03-22:
the other inverters sending their data via mqtt. With NodeRed they are received and processed. I just added some nodes calculating the values and sending them on another topic to the meterproxy

cratoo, 2025-08-01:
There is a way to say for sure:
check https://monitoring.solaredge.com/solaredge-web/p/site/<yoursiteid>/#/charts and add the energy produced for the external production meter.
For power the graph works for me flawlessly, but energy stays at 0, though I report values for energy_active and e.g. l3_energy_active via semp-tcp.py

macruch2, 2025-08-01:
I have the same result in the chart. I have changed the scaling kwh to wh. Tomorrow I'll see, if there has changed something.

cratoo, 2025-08-02:
@macruch2 yesterday I finally was able to successfully report energy to solaredge monitoring portal. Here is what I did/finally found in other posts:
1. In your guide above you already mentioned that meter type should be "WattNode". Mine was solaredge all the time.
2. In https://github.com/nmakel/solaredge_meterproxy/pull/30#issuecomment-2083340336 I found the the comment of sicheste  that he/she had this running for some time now, so I checked [his deye code](https://github.com/sicheste/solaredge_meterproxy/blob/add_deyemicro/devices/mqttDeyeMicro.py)
3. The code publishes "energy_active", "l1_energy_active", "import_energy_active", "l1_import_energy_active" via mqtt to rs485. Until now I was missing (l1_)import_energy_active, so I added both of them.
4. 15 minutes later 0.1 kWh appeared on the monitoring.
5. As nmakel posted it somewhere else, the unit is kWh, so for 100Wh you need to publish 0.1 to above metrics, and also the number is cumulative. So if you have another 100Wh, you'd post 0.2 so that the published number always only increases.

For clarity, currently I post the following to mqtt, from which meterproxy picks it up:
`{"l1_power_active":262.86,"l2_power_active":260.87,"l3_power_active":489.14,"l3_energy_active":1.607,"l1_energy_active":0,"l2_energy_active":0,"energy_active":1.607,"l3_import_energy_active":1.607,"import_energy_active":1.607}`
And yes currently L1+L2 energy are missing in this, but for L3 this works.
Personally I find it counter intuitive that one has to publish solar production energy to a field named import_energy_active, but as long as it works...
Hope this helps you too!

macruch2, 2025-08-04:
the day is not yet over, but the data looks fine. Also the consumption data are shown reasonable in the app. Without the extra meter the consumption data have decreased during the day when the external inverters have produced more power than needed. 

I'll update the documentation about this.